Ultimaker/Cura #5956: CURA v4.1.0 Fullscreen View toggle

Twopusses, 2019-07-01:
**Application version**
CURA 4.1.0

**Platform**
Windows 10/AMD Radeon 7970m

**Printer**
CTC i3

**Reproduction steps**
I accidentally hit F11 

**Actual results**
 F11 apparently toggles FULL SCREEN VIEW but that left me with 'disabled' menu headers so I couldn't see how to UNDO this. 

**Expected results**
Since I accidentally came across the feature, I can only say that F11 did what it was intended to do BUT I'd have expected it to be printed somewhere in the screen HOW TO REVERSE F11 Change to Full Screen View otherwise you're left hanging like a plum : ) I had to close my program to get out of this and to my way of thinking that's either a BUG or 'bad' programming, but I just want to quickly say, I LOVE Cura. As a piece of software it's brilliant., and yes, this is my first ever Bug Report.

**Additional information**
 I don't know what LOG FILES are in this context or how to obtain them. If you send details to me I'm sure I can locate what you need.

mbartlett21, 2019-07-02:
@Twopusses 
You can hit F11 again to get out of that mode :)

Ghostkeeper, 2019-07-02:
I don't think we should print somewhere in the screen how to get out of F11. That would not look good for the 99% of users that do know how they got in F11. The application menu should show fine though. Here's how it looks on my computer:

[image]

The application menu is there at the top. Is it not for you? This could be platform-dependent.

mbartlett21, 2019-07-02:
@Ghostkeeper I am on Windows 10 1903 and I see the bar at the top, but no menus open up when I click on them.

diegopradogesto, 2019-07-02:
We tested on Windows and it works. When pressing F11 (or View > Toggle Full Screen) it goes to full screen mode and the menu shows up on top. However it seems that it's not working correctly. 

First thing I noticed is that it's laggy and sometimes the submenu doesn't appear, making this functionality a bit unusable. However there is the workaround of pressing F11 again and then the menus work correctly again.

Twopusses, 2019-07-03:
My Thanks to everyone who's pointed out that pressing F11 again will undo the action, but it remains true that if you don't know how you got into Full Screen Mode, having to systematically press "every key/combination on your keyboard" trying to find out which one might achieve that, and in so doing you may crash your machine, close CURA or lose your file etc. this isn't a good 'resolution'. It is "a Bug" albeit one of a hidden knowledge issue than strictly a coding flaw although it seems others have spotted other issues associated with this functionality. I agree that printing "F11 for Original Screen Mode" would be clumsy/untidy. Thanks again for your help & attention with my first Github Bug Report.

mbartlett21, 2019-07-03:
Maybe a popup could be added that fades away after a few seconds (like fullscreen on chrome)

Ghostkeeper, 2019-07-03:
Oops. I didn't mean to close this via the commit. Sorry.

We've discussed this with the team, and agreed that adding Escape as hotkey to exit full screen should make it easier to discover.

The issue with the application menus not working while in full screen mode is still open, but we haven't been able to reproduce that problem. The menus are working fine here, except that the Settings menu is a bit slow (because it needs to build that model upon opening it, which takes a second).

rburema, 2019-07-05:
While some of us can reproduce the problem now (but others on the same OS are fine), we've decided to defer the rest of this issue, sorry.

commit 365d725e (2019-07-03): Add escape as hotkey to exit full screen

Should be easier to discover.

Fixes #5956.
closed 2023-06-13 by github-actions[bot]