Ultimaker/Cura #5958: [4.1.0] Backend stuck on slicing - only on windows

nickbrandwood, 2019-07-02:
**Application version**

4.1 

**Platform**

Windows 

**Printer**

Anycubic I3 Mega

**Reproduction steps**

Add file, click slice.
Slicing always stops, same issues on two different Windows machines. Macbook slices same projects with no issues. 

**Actual results**

it get stuck on slicing. The backend is able to connect, but at a certain point it will get stuck with:
2019-07-02 09:29:49,685 - DEBUG - [Thread-212] UM.Backend.Backend._backendLog [101]: [Backend] [ERROR] Could not send message header

Tried running both with and without Administrator.

**Expected results**

Expect that it finishes slicing.

**Additional information**

[cura.log](https://github.com/Ultimaker/Cura/files/3348826/cura.log)
[AI3M_TileBox.zip](https://github.com/Ultimaker/Cura/files/3348833/AI3M_TileBox.zip)

Ghostkeeper, 2019-07-02:
Hmm, cannot reproduce the issue. It slices fine on this side:

[image]

Is this happening with all models or just this one?

The error "could not send message header" seems to indicate that something is wrong with the local socket connection between Cura's front-end and the slicing engine. Maybe some firewall is blocking that?

nickbrandwood, 2019-07-02:
all models. I thought that it might be a firewall issue too, but then I noticed that the backend seems to converse correctly until this line:

2019-07-02 11:37:59,170 - DEBUG - [Thread-216] UM.Backend.Backend._backendLog [101]: [Backend] G5" -s support_xy_overrides_z="z_overrides_xy" -s support_skip_zag_per_mm="20" -s meshfix_maximum_travel_resolution="0.8" -s wireframe_bottom_delay="0" -s support_bottom_height="1" -s day="Tue" -s material_bed_temperature_layer_0="60" -s machine_max_acceleration_e="10000" -s wall_line_width_x="0.4" -s machine_minimum_feedrate="0.0" -s machine_name="Anycubic i3 Mega" -s raft_interface_line_width="0.8" -s z_seam_type="sharpest_corner" -s skin_line_width="0.4" -s raft_interface_jerk="8" -s wall_0_wipe_dist="0.2" -s skirt_line_count="3" -s filter_out_tiny_gaps="True" -s support_tree_branch_diameter_angle="5" -s raft_surface_line_spacing="0.4" -s speed_support_bottom="20.0" -s support_extruder_nr_layer_0="0" -s bridge_skin_material_flow="60" -s speed_travel_layer_0="50.0" -s machine_extruder_count="1" -s ironing_enabled="False" -s speed_support_infill="30" -s retract_at_layer_change="False" -s ooze_shield_dist="2" -s ironing_flow="10.0" -s infill_line_width="0.4" -s fill_outline_gaps="False" -s meshfix_union_all="True" -s machine_max_jerk_z="0.4" -s retraction_amount="6" -s z_seam_corner="z_seam_corner_inner" -s machine_nozzle_size="0.4" -s material_flow_layer_0="100" -s jerk_topbottom="8" -s minimum_support_area="0.0" -s switch_extruder_retraction_speed="20" -s infill_offset_y="0" -s roofing_angles="[ ]" -s layer_start_x="0.0" -s meshfix_maximum_deviation="0.05" -s wireframe_straight_before_down="20" -s acceleration_print="1800" -s support_tree_wall_thickness="0.4" -s support_brim_width="8.0" -s material="0" -s support_roof_pattern="concentric" -s infill_sparse_density="25" -s retraction_speed="40" -s support_roof_height="1" -s connect_skin_polygons="False" -s wireframe_roof_inset="3" -s cool_min_layer_time="5" -s support_tower_diameter="3.0" -s meshfix_maximum_resolution="0.5" -s machine_max_feedrate_y="299792458000" -s meshfix="0" -s spaghetti_infill_enabled="False" -s acceleration_travel_layer_0="3000.0" -s bridge_fan_speed_2="0" -s speed_wall="50" -s wireframe_height="3" -s cool_fan_speed_max="100" -s support_mesh_drop_down="True" -s prime_blob_enable="False" -s extruder_prime_pos_y="0" -s bottom_skin_expand_distance="1.2000000000000002" -s wall_thickness="1.2" -s machine_height="205" -s min_infill_area="0" -s prime_tower_flow="100" -s raft_interface_thickness="0.30000000000000004" -s ooze_shield_angle="60" -s cool_lift_head="False" -s wireframe_strategy="compensate" -s coasting_volume="0.064" -s gradual_support_infill_steps="0" -s bottom_thickness="1.2" -s top_bottom_pattern="lines" -s machine_nozzle_expansion_angle="45" -s support_type="everywhere" -s bridge_fan_speed="100" -s skirt_gap="5" -s speed_prime_tower="50" -s machine_endstop_positive_direction_z="True" -s support_brim_line_count="15" -s infill="0" -s wipe_hop_amount="1" -s support_pattern="triangles" -s bottom_skin_preshrink="1.2000000000000002" -s raft_surface_thickness="0.2" -s support_xy_distance_overhang="0.2" -s retraction_min_travel="1.5" -s material_flow_temp_graph="[[3.5,200],[7.0,240]]" -s travel="0" -s infill_extruder_nr="-1" -s jerk_wall_x="8" -s magic_fuzzy_skin_thickness="0.3" -s support_initial_layer_line_distance="8.0" -s jerk_support_bottom="8" -s remove_empty_first_layers="True" -s support_join_distance="2.0" -s machine_steps_per_mm_z="50" -s wireframe_top_jump="0.6" -s infill_before_walls="True" -s wipe_retraction_retract_speed="40" -s retraction_enable="True" -s skin_outline_count="1" -e0 -s acceleration_wall_0="1800" -s cool_fan_speed_min="100" -s support_fan_enable="False" -s prime_tower_wipe_enabled="True" -s support_angle="60" -s meshfix_maximum_deviation="0.05" -s support_interface_density="100" -s support_roof_density="100" -s top_thickness="1.2" -s jerk_wall="8" -s ironing_only_highest_layer="False" -s support_interface_pattern="concentric" -s acceleration_skirt_brim="1800" -s material_print_temperature_layer_0="0" -s skirt_brim_minimal_length="250" -s infill_angles="[ ]" -s machine_extruder_end_pos_y="0" -s support_conical_angle="30" -s raft_surface_jerk="8" -s multiple_mesh_overlap="0.15" -s limit_support_retractions="True" -s acceleration_travel="3000" -s retraction_hop_only_when_collides="True" -s retraction_hop_after_extruder_switch="True" -s raft_base_line_spacing="1.6" -s machine_steps_per_mm_z="50" -s speed_support_roof="20.0" -s support_infill_rate="15" -s support_roof_offset="0.0" -s speed_wall_x="50" -s support_brim_enable="False" -s raft_base_acceleration="1800" -s command_line_settings="0" -s switch_extruder_prime_speed="20" -s retraction_combing_max_distance="0" -s support_bottom_stair_step_height="0.3" -s support="0" -s acceleration_ironing="1800" -s extruder_prime_pos_z="0" -s skin_preshrink="1.2000000000000002" -s machine_feeder_wheel_diameter="10.0" -s wall_0_inset="0" -s support_interface_skip_height="0.3" -s machine_heat_zone_length="16" -s mesh_position_z="0" -s machine_extruder_end_pos_abs="False" -s travel_compensate_overlapping_walls_enabled="True" -s wipe_retraction_prime_speed="40" -s skirt_brim_line_width="0.4" -s support_interface_line_width="0.4" -s support_minimal_diameter="3.0" -s center_object="False" -s raft_smoothing="5" -s wall_min_flow="0" -s wipe_hop_enable="True" -s extruder_nr="0" -s switch_extruder_retraction_speed="20" -s jerk_layer_0=" [... cut ...] interface_line_width="0.8" -s acceleration_support_bottom="1800" -s roofing_pattern="lines" -s raft_surface_layers="2" -s prime_tower_min_volume="6" -s speed_support_bottom="20.0" -s material_guid="0ff92885-617b-4144-a03c-9989872454bc" -s raft_base_jerk="8" -s raft_speed="25.0" -s z_seam_x="105.0" -s raft_surface_speed="25.0" -s acceleration_support_roof="1800" -s blackmagic="0" -g -e0 -l "0" -s extruder_nr="0"

and then it stops, so it seems to me more that the backend is crashing on something.

Ghostkeeper, 2019-07-03:
Could be related to a setting. Can you provide us with a project file? The 3MF file you attached earlier is plain 3MF, not a project file. You can save a project file by going to File -> Save...

seth2371, 2019-07-03:
I am having the same issue. It stops slicing halfway through (the loading bar stops and nothing happens). Worked fine in the morning then stopped working in the afternoon. The only change was the diameter of the prime tower. I have restarted and reset everything I could find. 
[UMS5_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/3355150/UMS5_3DBenchy.zip)

Ghostkeeper, 2019-07-04:
Seth2371's issue is different though. It doesn't say anything about sending message headers in the log. Also, that one actually reproduces for me as well.

Ghostkeeper, 2019-07-04:
I fixed that infinite slicing issue for Seth2371's model. Should be fixed in version 4.2. The original issue still stands though. We haven't been able to reproduce it yet.

Vandrasc, 2019-07-05:
Hi @nickbrandwood , could you please share a project file with us so we can try to reproduce the issue reported by you? Unfortunately, I am also not able to reproduce using just your model. Thanks!

nickbrandwood, 2019-07-05:
sorry about the delay in sending this, I was away for a few days.
[AI3M_TileBox.zip](https://github.com/Ultimaker/Cura/files/3361736/AI3M_TileBox.zip)

nickbrandwood, 2019-07-05:
I opened that file on a Macbook and it worked sliced fine as well as on my other widows machine that has started slicing too, though I don't know what changed. If no one else is having this issue I think its safe to say that its something in the local environment of the specific machine - I'm happy to close the issue.

Vandrasc, 2019-07-05:
No worries, thank you for sharing! 
The slicing is also completed for me, so maybe we could indeed close this ticket and re-open if it happens again.  At least one is fixed :) 

Thank you both for your feedback!

closed 2019-07-05 by Vandrasc

TimetravelerDD, 2019-12-22:
I can confirm this is still an issue with cura 4.4 on windows

I read in many threads that the windows firewall is the culprit. For me it was not the issue, but I can not not slice anything when my VPN is on. I am using ExpressVPN in split tunnel mode (not even active for cura). I can reproduce it every time even after re installation and clearing of settings. There is definitely some weird network thing going on with the cura slicer.

Ghostkeeper, 2019-12-23:
Cura starts CuraEngine as a separate subprocess and communicates to it using a local socket (127.0.0.1:49674, incrementing the port number every time there is a communication error). Some antivirus software suites pick up on this, not trusting a local port. Local ports are very common in Linux software but not so common in Windows, so virus scanners sometimes overzealously seem to think that Linux software is always hacker software and block it. That's why you may have read that antivirus software  can block Cura.

Your VPN client may also do something with routing local ports. A copy of your Cura log may shine some light on what's happening. If it's the VPN software I'd expect it to get a communication error. If it's an antivirus then I'd expect that it can't start CuraEngine at all and give something like a permission error or just a generic OS error.