Ultimaker/Cura #5962: GCODE Meta features

aryez91, 2019-07-02:
**Is your feature request related to a problem? Please describe.**

I'm always frustrated when I have tons of unusable gcodes accumulating after few months/years.
They are unusable prffesionaly because I don't remember what settings are in those files. Physical size of model, estimated filament, time, quality settings...

**Describe the solution you'd like**
It would be great if cura could write meta data to the gcode.
This could include basic print settings (quality, build plate adhesion, supports) a non default applied settings, and physical properties (size, number of models etc)...

Secondly, it would be nice if cura asked on next launch or something if the printing ended successfully, and  include this new meta information on the source gcode.

**Describe alternatives you've considered**

Today I manually include some settings in the filename: filament required and printing time,
And it's ok. But the filename would get long if I tried to include all meaningful info, and it's also manual labor.

Another partial alternative is to preview the gcode, but that usually takes a long time and doesn't recompute filament required and time.

**Affected users and/or printers**
 
People who have clients that ask for reprints of old model.
Anyone who wants to quickly reprint an older file, but want to double check the settings, differentiate variations.
[image]
 

**Additional context**

Ghostkeeper, 2019-07-03:
The settings you've changed already get appended to the g-code. It's at the very bottom of the g-code. The model dimensions also get written to the g-code header (in 4.1 only in the Griffin flavour, but since 4.2 with every flavour). However there is no shell integration as you've shown in that screenshot. Cura can import the g-code as a profile I believe though.

Ghostkeeper, 2019-07-03:
Getting feedback on how good the print went is a separate issue. I once made a prototype that did this here: https://github.com/Ghostkeeper/R2D2 This added a new stage after Monitor where you could fill in how good the print went and it would store this in Cura's configuration folder (rather than in the g-code because the g-code file might not exist at all since we sent the print via USB or wifi, or maybe the file moved, etc.) In my case that feedback was used to tweak your settings better but it could also just be used as reference. In any case, it didn't come through. I don't think we'll implement that feedback tracking unless we find a stronger need for it.

GregValiant, 2023-09-15:
See PR #15854 It's in line for review.
Can this be closed?

closed 2023-09-29 by github-actions[bot]