Ultimaker/Cura #5964: Build plate temperature reverts to 110C

wcray, 2019-07-03:
Modifications to the build-plate temperature are not respected in saved profiles, immediately revert to 110deg C when the profile is saved, or when "update profile with changes" is selected.

**Application Version**
Cura-Lulzbot 3.6.8, FlexyDually mode

**Platform**
Mac-OSX 10.11.6

**Printer**
Taz-6, FlexyDually

**Steps to Reproduce**
Change the build plate temperature to anything you like.  Select "Update Profile with current changes/overrides".  

**Actual Results**
Build plate temperature reverts to 110 deg C.

**Expected results**
Build plate temperature remains at temperature it was changed to.

**Additional Information**
This happens even if the build-plate temperature is set to the same (other-than-110deg) temperature under both extruders.  If it's set to different temperatures, Cura knows and the info-pop-up says the value will be resolved from the differing temps.  Cura then ignores this and sets it back to 110C immediately upon saving the profile.

Cura actually /is/ obeying the displayed value.  If a model is sliced and saved to gcode before saving the profile, the gcode will contain the correct (not-110deg) build plate temperature.  If the profile is saved and the field reverts to 110 however, the gcode will reflect the 110deg temperature instead.

It's possible this only happens with the PLA(Verbatim) material profile and custom materials derived from it.  I haven't tried others.

Sure wish you wrote you configuration files in XML/plain-English, so that we could just tweak this stuff directly in the definitions!

fieldOfView, 2019-07-03:
Unless you can confirm the same thing happens in Ultimaker Cura 4.1, there is very little Ultimaker can/will do about this. Lulzbot made a lot of changes to their version of Cura, especially in the materials and profile handling, so chances are they introduced this bug. Please report this issue in their tracker at http://code.alephobjects.com

wcray, 2019-07-03:
I rather figured that.  This seems the most active community for general discussion of Cura problems, and so I thought that cataloging the issue here might help someone in the future, even if it wasn't likely to be addressed by Ultimaker.

Doing a bit more digging myself after posting the issue, I believe I've isolated the problem to what I believe is an "undocumented feature" in the (at least Cura-lulzbot) software:   Overrides/customizations that are applied to "an extruder", that are universal (like the build plate temp), are applied as overrides/customizations across the settings for all extruders.  However, when the overrides/customizations are saved for "an extruder", any competing values for universal settings that appear in the _profiles_ for the other extruders, come back and re-populate the settings in the extruder you're working on.

The existence or non-existence of this same logic fuddle should be easy for anyone with the Ultimaker version of the software to confirm:  Set up a build with a PLA profile in the first extruder and an ABS profile in the 2nd extruder.  Customize the build-plate settings for the PLA extruder and save a custom profile for your modified PLA settings.  Change the build-plate settings again.  Go to the 2nd extruder to confirm that the build plate setting customizations are the same for both extruders. Go back to the first extruder, and update the custom profile with the new changes.  If your changes stick, and gcode written now contains the build-plate settings you just entered, woo-hoo, you're bug-free!

fieldOfView, 2019-07-03:
The LE version of Cura should really be considered as a different application; a lot has been changed by the Alepha Objects team. Reporting or discussing issues with that version here will only further confuse matters.

nallath, 2019-07-03:
I've just tried to reproduce this but this doesn't happen for Cura. I guess that the lulz-bot SW engineers did something that broke that profile logic. 

That being said; Although that they use a similar version system as Ultimaker, their version actually branched off around 3.2, which means they are pretty far away from this code base (They don't have the changes included in 3.3, 3.4, 3.5, 3.6, 4.0 and 4.1). 

As such, I'd recommend that you contact them.

closed 2019-07-03 by nallath