Ultimaker/Cura #5968: Backup Errors

HACKER-3000, 2019-07-03:
**Application version**

**Platform**

Windows 10

**Printer**

Ender 3

**Reproduction steps**

Do a Backup and try to restore it!

**Actual results**

Config Corrupted, and parts not even there!

**Expected results**

Fine restore of Backup.

**Additional information**

A screenshot of the disaster:
[image]

nallath, 2019-07-03:
Is this the case with all backups? Was this already the case when the backup was done? Could you share your configuration folder and logs?

HACKER-3000, 2019-07-04:
yep it is with all and allready happend once a few months ago... Also i tried to restore it(This backup from 03.07.19) just a few hours after createing it because i screwd the configuradion of visible settings...

Heres my cura folder: http://hacker3000-data.bplaced.net/cura.zip

Ghostkeeper, 2019-07-05:
Hmm, that Cura folder is not downloading for me.

As a workaround, you can try restoring just the preferences file (`cura.cfg`) from the zip. That preferences file contains the list of visible settings.

I just tried the back-up system and I'm not getting any problems with restoring it. No configuration errors.

HACKER-3000, 2019-07-05:
its just SLOW AF

HACKER-3000, 2019-07-05:
new link: https://hacker3000.goip.de/cura.zip

Ghostkeeper, 2019-07-08:
Hmm, well I see one thing wrong with Cura here. Loading up that folder causes me to log in successfully under your name. Since the cura.cfg file contains a log-in token I can access your account from here. You'd best take that zip file down if you can (I've got a copy, but I promise I won't do anything weird).

We should probably store that log-in token somewhere else, or encrypt it with an xor of your CPU ID or something so that it can't be shared...

Ghostkeeper, 2019-07-08:
It's not the back-up plug-in. Your original configuration folder has the same error. The file is missing that contains the changes that were not saved to a profile.

I don't know how your configuration folder got into that state. I'm unable to reproduce that except by copying your configuration folder in. Perhaps your configuration got into that state via faulty back-ups at some point? It's hard to tell, now.

HACKER-3000, 2019-07-08:
@Ghostkeeper I allready expected sth like this with the Login...
Btw the idea with the cpu id xor isnt that bad!

HACKER-3000, 2019-07-08:
and ive removed the file and replaced it with a html saying that is was removed

closed 2020-06-19 by no-response[bot]