Ultimaker/Cura #5969: Make Color Scheme that shows changes made to model through "Make Overhang Printable"

usshadowop, 2019-07-03:
**Is your feature request related to a problem? Please describe.**
Not able to see everywhere the model has been modified when "Make Overhang Printable" is enabled. 

**Describe the solution you'd like**
It would be nice if there was a Color Scheme that allowed you to see in a separate color anywhere material was added in order to make overhangs printable so you could quickly see all the locations that have been modified in your model.

**Describe alternatives you've considered**
Comparing side by side screenshots to look for any deviations and identify changes visually.

**Affected users and/or printers**
Anyone that utilizes the "Make Overhang Printable" feature.

**Additional context**
[image]

Ghostkeeper, 2019-07-04:
Hmm, yeah that would make the feature easier to understand. However the implementation of this is going to be difficult, because "Make Overhang Printable" modifies the outlines of the mesh; one polygon becomes another one. Part of this polygon's outline gets changed. Then we do a gazillion different things to the polygon, like horizontal offsets, insets to generate the various walls, reducing the resolution, etc. And after all that, part of the insets for the outer walls need to be marked differently. We don't currently store anything per-part-of-a-polygon so that'd mean that the walls need to be broken up into multiple polylines.

In short, I don't think we'll be giving this a try any time soon. If anyone can find an easier way to implement this, please show us!

usshadowop, 2019-07-05:
So this is coming from someone who hasn't done much work with 3d object types or has much knowledge of how the data is stored / manipulated so this may be a dumb suggestion.

But would there be a way of taking a copy of the stl prior to slicing, then taking the model after the manipulations are made and comparing them? Effectively is there some method for doing a "diff" or a comparison of the two and seeing how the geometry varies (original A / and overhang modified B). May not be able to color easily from this info but at least may be possible to have an idea of whats been changed easily using this comparison.

Ghostkeeper, 2019-07-05:
That can be done in a way yes. There are two ways I can think of to do this but both are significantly flawed.

1. We could analytically calculate the 3D distance from one 3D model to another. This is extremely complex though, and if we get it working I'd expect that this calculation will take a good hour or two on a reasonably sized g-code. One thing that we'd also need to generate is a 3D model of the g-code, which is currently only generated in the GPU. This model will have self-intersections which are going to need to be removed for many algorithms to work correctly on 3D models. Removing these self-intersections on a generic surface mesh (rather than volumetric) is an unsolved problem in science at the moment so we'd need to stay with volumes (cylinders) which is going to require us to write a whole new system to deal with volumes.
2. You could use the renderer to show the difference. The GPU creates a depth buffer which is normally used to test which triangles are in front of the other ones. You can access this depth buffer and colour stuff differently on the screen if the difference between two depth buffers. This is probably fast (reduces frame rate a bit perhaps) but it will look different when looking from different angles. In particular, when looking at a normal surface from a shallow angle, you'd see great difference between the original mesh and the g-code because the place where the g-code is behind the surface is much farther then. Another problem with this solution is that the g-code is sometimes smaller for other reasons as well (like minimum thickness due to line width) and sometimes bigger (horizontal expansion, or even support).

Both of these solutions will also suffer from the problem that the g-code's surface is ribbed. So you'd see that there is a difference between model and g-code because the g-code looks thinner in between the layers.

usshadowop, 2019-07-05:
Thank you for the awesome explanations of the two methods detailed above, definitely makes a lot more sense now why this is incredibly more complex than it would seem on the surface.

closed 2023-07-13 by github-actions[bot]