Ultimaker/Cura #5970: Cura stuck on welcome screen on new install at Manjaro linux

dreanmer, 2019-07-03:
**Application version**
4.1.0

**Platform**
Manjaro 18.0.4 (Arch linux)

**Printer**
Tronxy X5S

**Reproduction steps**
Install cura 4.1.0 via `sudo pacman -Syu cura` 

**Actual results**
Stuck on welcome screen, nothing work clicking or typing any key.
[image]

**Expected results**
Close welcome overlay

**Additional information**
Shortcuts still working (eg.: open file) but the operations dont work at all.

Ghostkeeper, 2019-07-04:
Are you working with the AUR package, or did you use the AppImage? If you were using the AUR package, does the AppImage work properly?

We've seen before that installing the AUR package causes Cura to use a more modern version of Qt, which breaks Cura. Qt should(tm) not break backwards compatibility in minor releases, but maybe they have or maybe Cura is somehow relying on bugs that were fixed in newer versions.

dreanmer, 2019-07-05:
No, its the community version, from here: https://www.archlinux.org/packages/community/any/cura/.
BTW `pacman -Syu` update all packages on the machine before install the chosen package.

szma, 2019-07-07:
I have the same issue on arch linux.

b-nogal, 2019-07-08:
Have same problem. Running with sudo "fixes" it but its not really a fix...

Ghostkeeper, 2019-07-08:
You're going to need to report this here: https://bugs.archlinux.org/?project=5&string=cura

We don't maintain that package. We can't really do anything about it.

shawe, 2020-04-10:
Same problem here, and solved with this command:

> env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP cura

Read it here https://bugs.archlinux.org/task/62133

cipreses, 2020-09-15:
I put this sentence in the launcher:

env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /usr/bin/cura %F 

IWFM Thanks

Ghostkeeper, 2020-12-04:
We're opening this issue again, because supporting Qt 5.15 is going to be brought into scope soon. We have plans to upgrade the Qt version in our own distribution to 5.15.

Neopect, 2020-12-15:
I also just recently installed `Manjaro` and `Cura` and been having the same issue, text missaligned with no clickable areas. I was able to get past part of the setup by smashing `Tab` & `Enter` but im still stuck on the data collection question tab.

My versions:

- Manjaro 5.9.11-3
- KDE Plasma 5.20.4
- Qt 5.15.2
- Cura 4.7.1-1

csivanich, 2020-12-27:
Currently seeing issue in Garuda as well:
[process output - cura.log](https://github.com/Ultimaker/Cura/files/5743940/cura.log)
[image]
Kernel/packages:
```
 c@muni-xps13 ~: uname -a; for i in cura kwin qt; do pacman -Qs $i; done | grep local | sort | uniq  | column -t 
Linux muni-xps13 5.10.2-104-tkg-bmq #1 TKG SMP PREEMPT Thu, 24 Dec 2020 00:59:51 +0000 x86_64 GNU/Linux
local/appstream-qt                        0.13.1-1
local/attica                              5.77.0-1                (kf5)
local/avahi                               0.8+15+ge8a3dd0-3
local/bluez-qt                            5.77.0-1                (kf5)
local/cura                                4.7.1-2
local/cura-binary-data                    4.7.1-1
local/curaengine                          4.7.1-2
local/cura-resources-materials            4.7.1-1
local/karchive                            5.77.0-1                (kf5)
local/kcolorpicker                        0.1.4-3
local/kcoreaddons                         5.77.0-1                (kf5)
local/kdbusaddons                         5.77.0-1                (kf5)
local/kdsoap                              1.10.0-1
local/kguiaddons                          5.77.0-1                (kf5)
local/kirigami2                           5.77.0-1                (kf5)
local/kitemmodels                         5.77.0-1                (kf5)
local/kitemviews                          5.77.0-1                (kf5)
local/kquickcharts                        5.77.0-1                (kf5)
local/kvantum-qt5                         0.17.0-1
local/kwayland                            5.77.0-1                (kf5)
local/kwidgetsaddons                      5.77.0-1                (kf5)
local/kwin                                5.20.4-1                (plasma)
local/kwindowsystem                       5.77.0-1                (kf5)
local/kwin-effects-sliding-notifications  1.0.3-1
local/kwin-scripts-forceblur              0.4.1-1
local/libaccounts-qt                      1.16-2
local/libdbusmenu-qt5                     0.9.3+16.04.20160218-5
local/libinput_gestures_qt                0.3.2+7+g0868859-1
local/modemmanager-qt                     5.77.0-1                (kf5)
local/networkmanager-qt                   5.77.0-1                (kf5)
local/phonon-qt5                          4.11.1-2
local/phonon-qt5-gstreamer                4.10.0-2
local/plasma-framework                    5.77.0-2                (kf5)
local/plasma-integration                  5.20.4-1                (plasma)
local/polkit-qt5                          0.113.0-2
local/poppler-qt5                         20.12.1-1
local/pulseaudio-qt                       1.2-2
local/python-pyqt5                        5.15.2-2                (pyqt5)
local/python-pyqt5-sip                    12.8.1-3
local/qca                                 2.3.1-4
local/qqc2-desktop-style                  5.77.0-1                (kf5)
local/qt5-base                            5.15.2-2                (qt       qt5)
local/qt5-declarative                     5.15.2-1                (qt       qt5)
local/qt5-graphicaleffects                5.15.2-1                (qt       qt5)
local/qt5-location                        5.15.2-2                (qt       qt5)
local/qt5-multimedia                      5.15.2-1                (qt       qt5)
local/qt5-quickcontrols2                  5.15.2-1                (qt       qt5)
local/qt5-quickcontrols                   5.15.2-1                (qt       qt5)
local/qt5-script                          5.15.2-1                (qt       qt5)
local/qt5-sensors                         5.15.2-1                (qt       qt5)
local/qt5-speech                          5.15.2-1                (qt       qt5)
local/qt5-svg                             5.15.2-1                (qt       qt5)
local/qt5-tools                           5.15.2-1                (qt       qt5)
local/qt5-wayland                         5.15.2-1                (qt       qt5)
local/qt5-webchannel                      5.15.2-1                (qt       qt5)
local/qt5-webengine                       5.15.2-3                (qt       qt5)
local/qt5-x11extras                       5.15.2-1                (qt       qt5)
local/qwikaccess-git                      0.0.1.r65.ae84897-1
local/signond                             8.60-3
local/sonnet                              5.77.0-1                (kf5)
local/xdg-desktop-portal-kde              5.20.4-1                (plasma)
local/zbar                                0.23.1-5
```

LukasMueller187, 2021-01-17:
I am having the same issue on arch with kde plasma.

greppp, 2021-02-08:
Also having the same issue. Manjaro with KDE plasma.

thamudi, 2021-02-25:
> I put this sentence in the launcher:
> 
> env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /usr/bin/cura %F
>
> IWFM Thanks

This solution fixed it for me on manjaro 5.9, KDE 5.20,QT 5.15

jarrard, 2021-02-26:
> This solution fixed it for me on manjaro 5.9, KDE 5.20,QT 5.15

Ditto

closed 2021-10-07 by Ghostkeeper