Ultimaker/Cura #5972: workspace vanished

blindwolfcosplay, 2019-07-04:
ive encountered an issue that is recurring more and more recently. I open up cura 4.0.0 and the workspace just vanishes, I can load in an object and can even slice it. I just cant see it I have a blank space where the workspace should be. 

[image]

mbartlett21, 2019-07-04:
@blindwolfcosplay 
Can you try clicking on one of the view buttons in the bottom left?

blindwolfcosplay, 2019-07-04:
I can click on them. as well as clicking the same things up in the view menu on the top bar. It doesn't do anything,  or doesn't seem to as nothing changes in the viewport

closed 2020-06-19 by no-response[bot]