Ultimaker/Cura #5973: Skip adding a printer for restoring from backup

thecomputerguythatdoesntsuck, 2019-07-04:
**Is your feature request related to a problem? Please describe.**
It is slightly irritating when I have to add a new printer in Cura when I just want to restore from a backup during the initial setup. 

**Describe the solution you'd like**
Possibly sign into my cura account to restore a backup. Have this option before the add a printer during the initial setup. 

**Describe alternatives you've considered**
I couldn't come up with any. 

**Affected users and/or printers**
Anyone who is installing Cura on a new machine or new install due to hard drive failure, OS corruption, or that shiny new gaming computer. 
**Additional context**
N/A

Ghostkeeper, 2019-07-04:
In order to restore the back-up though, you'd have to be logged in. That logging in is currently at the end of the welcome screen for UX reasons.

HACKER-3000, 2019-07-08:
maybe a little link at the bottom of the first screen saying sth like "No, I want to load a backup"...

thecomputerguythatdoesntsuck, 2019-07-08:
That would probably be the best option.

diegopradogesto, 2019-07-10:
We have now a ticket in our backlog. For devs, please track CURA-6546