Ultimaker/Cura #5975: Brim corner clearance

StefanHamminga, 2019-07-04:
The brim feature works great for keeping a print attached to the build plate, but it can be time consuming to remove from some prints, especially around corners and curves. Part because it's not that easy to (quickly) cut around the corners or curves and part because inserting the knife isn't as easy as it can be.

For that reason I've resorted to including custom brims in the CAD models themselves, adding a bit of clearance to sharp corners. The created gaps don't seem to negatively impact the build plate adhesion, allow inserting a hobby knife a lot easier (and safer) and save a lot of tinkering cleaning up corners.

Of course it would be great if this could be added to Cura.

[image]

The above image shows a standard brim (left), a simple variant that should work in most cases (middle) and a bit more extensive variant that helps detaching the brim from long curved walls (right).

Ghostkeeper, 2019-07-04:
I'm worried that this makes the brim ineffective against warping. In those outside corners the brim is simultaneously the most important and the easiest to peel off, in my experience.

StefanHamminga, 2019-07-04:
So far no bad results in using this kind of brim. I normally clear max 2mm from the corner, but even less (0.5-1mm) would suffice I suppose. Of course I'm not expecting this to be a default, but it's been a good time saver for me (even modelling the cuts individually is faster than cleaning up prints, especially when iterating a design).

BagelOrb, 2021-11-23:
I kinda had the same idea. An intermittent brim could be a lot easier to remove. 

The idea is in some sense similar to the idea of mouse ears, but also kind of the opposite: https://github.com/Ultimaker/Cura/issues/1349

People don't seem to like brim at all because of the post-processing required, so they resort to using glues and the like.

I think glues also require pre-processing, so it doesn't really solve the issue.

I'm gonna do a test print and see how colleagues react to the idea.

guywire, 2022-01-28:
sounds like a good feature