Ultimaker/Cura #5976: Glare fix / matte preview

jfsmith343, 2019-07-04:
**Is your feature request related to a problem? Please describe.**

I'm very frustrated when I can't see details on my models do to glare in the preview.

**Describe the solution you'd like**

I would like to be able to change the surface of the object in the Cura preview to have a flat finish. This way people can see small details in the objects they are slicing.

**Describe alternatives you've considered**

**Affected users and/or printers**
  everyone
**Additional context**

[image]

Ghostkeeper, 2019-07-05:
With dark material colours, the glare is very necessary in order to show the shape of the model, because the shadows obviously won't do anything. We'd need it to adjust itself somehow to the brightness of the surface.

nubnubbud, 2019-07-16:
if you're using the default filament color, I would highly suggest you switch to dark mode. The lighting remains the same, but the part is shown over a dark grey background. I might, however, think that adding ambient occlusion would help greatly. I've come across this some, too. The issue is performance, for the most part. some parts have huge polycounts, and ambient occlusion is just one more pass that can slow it down. it wouldn't be a bad addition if it was optional, however...

Ghostkeeper, 2021-12-30:
In #11100 I proposed a couple of possible solutions to improve this:

* Add a second light source at an angle to draw specular reflections for. Maybe a third to get the classic 3 point lighting set-up.
* Implement ambient occlusion.
* Move the current light source to be at a greater angle away from the camera centreline.

Of those, I think ambient occlusion would be the most effective, but also the hardest to implement.