Ultimaker/Cura #5977: Preview colour bugs

mbartlett21, 2019-07-05:
**Application version**
Cura 4.1.0

**Platform**
Windows 10 1903

**Reproduction steps**
1. Load GCode below
2. Change preview mode to feedrate

**Actual results**
Lines after retractions fade from the retraction speed to their actual speed

**Expected results**
Lines after retractions show as their actual speed

**Additional information**
Screenshot:
[image]

<details>
<summary>
Example GCode (made with my own script on <a href="https://cad.onshape.com/documents/2e6cdddbacf401fc60fd2b91">Onshape</a>)
</summary>
<pre>;Start GCode
;Generated with Layers 2 gcode (https://cad.onshape.com/documents/2e6cdddbacf401fc60fd2b91)
;filamentDiameter 1.75
;lineWidth 0.4
;lineHeight 0.3
;filamentMMPer3dMM 0.0499
;minPathWithoutRetract 1.2
M140 S60  ;Heat bed to 60
M105      ;Report temperatures
M190 S60  ;Wait for bed temperature to reach 60
M104 S200 ;Heat hotend to 200
M105      ;Report temperature
M109 S200 ;Wait for hotend to heat
M82       ;Absolute extrusion mode
G28       ;Home
G1 Z15.0 F6000 ;Move the head down to 15mm above buildplate
;Prime the extruder
G92 E0    ;Reset the extruder value
G1 F200 E3 ;Extrude 3mm of filament
G92 E0
G92 E0
G92 E0
G1 F2100 E-7.5 ;Retract filament 7.5mm
M107 ;Fan off
G0 F3600 X-2.8685 Y-21.7882 Z0.3
G0 F2100 E0
G1 F700 X2.8685 Y-21.7882 E0.28622
G1 X8.4099 Y-20.3034 E0.57244
G1 X13.3783 Y-17.4349 E0.85865
G1 X17.4349 Y-13.3783 E1.14487
G1 X20.3034 Y-8.4099 E1.43109
G1 X21.7882 Y-2.8685 E1.71731
G1 X21.7882 Y2.8685 E2.00352
G1 X20.3034 Y8.4099 E2.28974
G1 X17.4349 Y13.3783 E2.57596
G1 X13.3783 Y17.4349 E2.86218
G1 X8.4099 Y20.3034 E3.14839
G1 X2.8685 Y21.7882 E3.43461
G1 X-2.8685 Y21.7882 E3.72083
G1 X-8.4099 Y20.3034 E4.00705
G1 X-13.3783 Y17.4349 E4.29326
G1 X-17.4349 Y13.3783 E4.57948
G1 X-20.3034 Y8.4099 E4.8657
G1 X-21.7882 Y2.8685 E5.15192
G1 X-21.7882 Y-2.8685 E5.43813
G1 X-20.3034 Y-8.4099 E5.72435
[... 558 lines cut ...]
G0 F7200 X-4.2066 Y-3.2
G1 F1050 X4.2066 Y-3.2 Z3.119 E118.65597
G0 F7200 X4.2066 Y-3.6
G1 F1050 X-4.2066 Y-3.6 Z4.381 E119.08041
G0 F7200 X-4.2066 Y-4
G1 F1050 X4.2066 Y-4 Z3.119 E119.50484
G0 F7200 X4.8 Y-4.8 Z3.03
G1 F1050 X-4.8 Y-4.8 Z4.47 E119.98915
G1 X-4.8 Y4.8 E120.46809
G1 X4.8 Y4.8 Z3.03 E120.9524
G1 X4.8 Y-4.8 E121.43134
G1 F2100 E113.93134
M140 S0 ;Cool heatbed
M107 ;Fan off
M104 S0 ;Cool head
M140 S0 ;Cool heatbed
;Retract the filament
G92 E1 ;Reset extruder value
G1 E-1 F300 ;Retract 2mm
G28 X0 Y0 ;Home axes
M84 ;Loosen motors
M82 ;absolute extrusion mode
M104 S0 ;Cool head
;End of gcode</pre>
</details>

Ghostkeeper, 2019-07-05:
This could be a problem of us only being able to render colours per vertex. Since g-code produces a gigantic amount of vertices to render, the beginning and ending of a line segment share the same vertex, and that vertex can only have one colour.
Am I right in this analysis, @nallath, or did we indeed duplicate all vertices at the borders? We must be doing something right in order to get the line width correct, but maybe we split up the polylines for different line widths.

This is not a problem for filament colour (since all polylines must get printed with one filament), line type (since connected lines always have the same line type and we do duplicate vertices for travel moves) or layer height (since all polylines belong to the same layer and thus have the same height).

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

mbartlett21, 2024-11-29:
It is fine with 5.9.0.
[image]

closed 2024-11-29 by mbartlett21