Ultimaker/Cura #5979: Slicing incorrect (and not constalt) when space between inner/outer walls is exactly times the nozzle size

LinoBarreca, 2019-07-05:
**Application version**
4.1.0

**Platform**
Win 7 x64

**Printer**
Ender 3

**Reproduction steps**
Started slicing a wall exactly wide 2mm (nozzle 0.4)

**Actual results**
vertical walls sometimes contain 5 lines sometimes 4 and fill between lines.

**Expected results**
Constant behavior through layers.

**Additional information**
Attached video.
The yellow part is because i have the "Shell - Fill gaps between walls" option active.
If I disable it there's a "hole" (missing line, which doesn't make sense) instead of the yellow filling (that wrong yellow filling increases a lot gcode size, printing time, head movement and vibrations during print)
[image]

[Video_2019-07-05_123245.zip](https://github.com/Ultimaker/Cura/files/3362288/Video_2019-07-05_123245.zip)

mbartlett21, 2019-07-05:
This seems like a duplicate of #4685

LinoBarreca, 2019-07-05:
It is similar but not the same. if you leave "fill gaps between walls" unchecked, the lines get "cut" in the middle of the wall for no reason at all.

mbartlett21, 2019-07-05:
Could you share the STL?

Ghostkeeper, 2019-07-08:
Rather a project file, because we'll need to have the exact same line widths to reproduce this issue.

LinoBarreca, 2019-07-08:
Attached.
Open, slice and go to layer 518 (for example)
[proto.zip](https://github.com/Ultimaker/Cura/files/3367115/proto.zip)

Ghostkeeper, 2019-07-08:
That's not a project file. That's just a 3MF file. To save a project file, you'd need to go to `File` -> `Save...`

As a result, the bug doesn't reproduce.

[image]

LinoBarreca, 2019-07-08:
Open it as a project. 
[image]

Ghostkeeper, 2019-07-08:
Ah, I see it now. In my log I get this error:

```
2019-07-08 11:01:26,763 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Could not check file file:///home/ruben/Desktop/proto.3mf
2019-07-08 11:01:26,765 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2019-07-08 11:01:26,766 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/CuraApplication.py", line 1751, in checkIsValidProjectFile
2019-07-08 11:01:26,766 - ERROR - [MainThread] UM.Logger.logException [88]:     result = workspace_reader.preRead(file_path, show_dialog=False)
2019-07-08 11:01:26,766 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/../plugins/3MFReader/ThreeMFWorkspaceReader.py", line 422, in preRead
2019-07-08 11:01:26,766 - ERROR - [MainThread] UM.Logger.logException [88]:     extruder_info.variant_info = instance_container_info_dict[variant_id]
2019-07-08 11:01:26,767 - ERROR - [MainThread] UM.Logger.logException [88]: KeyError: 'creality_ender3_0.4'
```

It then doesn't see this file as a valid project file and then just reads it as 3MF. The error indicates that the `creality_ender3_0.4` variant is missing. And that's correct; no such variant exists in Cura 4.1. Did you mod your Cura installation? Like with the CreawsomeMod?

LinoBarreca, 2019-07-08:
> It then doesn't see this file as a valid project file and then just reads it as 3MF. The error indicates that the `creality_ender3_0.4` variant is missing. And that's correct; no such variant exists in Cura 4.1. Did you mod your Cura installation? Like with the CreawsomeMod?

No. Standard installation. Windows x64. 
[image]

Ghostkeeper, 2019-07-08:
Looking at the model, I also have a suspicion of why it's inconsistent. If the wall thickness is exactly an odd multiple of the line width, it comes down to rounding errors. Cura works to a micrometre accuracy, and your mesh has a diagonal across the flat face (because it's a triangle mesh). The diagonal puts an extra vertex halfway the flat surface, which varies from layer to layer. If the coordinates of that vertex get rounded in one direction it could make the polygon disappear, but if it gets rounded in the other direction it could make the polygon appear again.

We have a partial fix for this in Cura 4.2 by making the resolution reduction algorithm remove vertices that have less than 5 microns deviation. That will probably fix it for this case but not for the general case, such as when the side of this model is sloped instead of perfectly vertical.

LinoBarreca, 2019-07-08:
I can understand the "your mesh has a diagonal across the flat face (because it's a triangle mesh). The diagonal puts an extra vertex halfway the flat surface, which varies from layer to layer." What I can't understand is why, if the side **is** vertical (I designed them myself in Inventor) the extra "vertex" is rounded.
[image]

Ghostkeeper, 2019-07-08:
The extra vertex is rounded because all vertices in CuraEngine are rounded to a 1 micrometre grid. In fact, the vertex in your original model is rounded to a grid as well (a grid with greater resolution towards 0 and lower resolution the further you get away from it) because all data on a computer has a finite resolution.

LinoBarreca, 2019-07-08:
> all data on a computer has a finite resolution.

I understand that there's finite resolution since I'm a developer.
I initially thought the line disappeared because you were using floating points for calculations and there were floating point rounding errors during calculations like:

do the math (probably using some divisions)
how much space is left on the right side of the vertex?
0,400000000000000001
ok a line fits, let's draw a line.
how much space there's on the other side?
0,39999999999999999.
A line doesn't fit so leave it empty (or fill the space if the Fill gaps between walls option is set)

Ghostkeeper, 2019-07-08:
> I initially thought the line disappeared because you were using floating points for calculations and there were floating point rounding errors during calculations like:

CuraEngine actually uses fixed point. This prevents rounding errors and makes things not behave differently if you move them to different places on the build plate.

> how much space is left on the right side of the vertex?

You can't really tell how much space is on one side of a vertex. This is working with polygons here, so a line might fit on one endpoint but not fit on the other. Instead, CuraEngine uses insets. It performs a polygon inset of half a line width from the original boundary to determine where the outer wall goes. If the polygon is in some place thinner than 1 line width, insetting by half a line width on both sides causes the polygon to disappear there. That's how Cura knows that a line doesn't fit any more.

> In this case, instead of removing the vertex, the "does the line fit" should have some extra tolerance.

This just moves the edge case by 1 micrometre. You'd get the same behaviour then if you were to have a model that is 1 micrometre wider.

LinoBarreca, 2019-07-08:
> This just moves the edge case by 1 micrometre. You'd get the same behaviour then if you were to have a model that is 1 micrometre wider.

it doesn't. because if the wall is, as in the project I sent, 2mm and the nozzle is 0.4 they must be 5 lines (by math) it would not have the same behavior because a wall thick 2,001mm is deliberately leaving 1 micrometer empty in the middle. this is by design, not a rounding error. the designer knows that he will have 2 outer lines, 3 inner lines (or vice-versa) and 1 micrometer gap.

if the error was due to floating point roundings you could have a rounding error in the 24th (or something) decimal. that error makes a IF comparison fail while it should (logically) succeed...but still the 24th decimal is a length way smaller than a micrometer.
Basically with floating point math the engine would do something like:
- inner wall1: 0.4
- inner wall2: 0.40000000000000000000000000000000000000000000000000000001,
- inner wall3: 0.39999999999999999999999999999999999999999999999999999999 NO LINE!
- outer wall2: 0.4
- outer wall1: 0.4

and in this case considering
4.00000000000000000000000000000000000000000000000000000001,
4 and
3.99999999999999999999999999999999999999999999999999999999
all the same (the lines would be all drawn)

as you said, though, if the engine uses fixed point math. I wonder how you could have that rounding on a plane which is completely vertical. there must be a bug somewhere, besides the extra vertex, that should be looked at.
Now you have the project, the model, the settings, and the source code. I guess you (or someone else) can debug it and see what happens and why. I don't think you need more information.

Ghostkeeper, 2019-07-09:
The project does open in Cura 4.1 (not in master, so we've discovered a bug there). Indeed if the outer wall line width is 0.399 instead of 0.4, the problem disappears.

> As you said, though, the engine uses fixed point math. I wonder how you could have that rounding on a plane which is completely vertical.

Take this example:
[image]
If there's a vertex on that black spot on the right hand side of the red line, it rounds towards the bottom right. If (on the next layer) there is a vertex on that black spot on the left hand side of the red line, it rounds to the top left. As a result, on the higher layer the distance to the other red line will be lower. It jitters by up to 2*sqrt(2) rounding margins.

closed 2019-07-09 by Ghostkeeper