Ultimaker/Cura #5980: Sloped Skin Layers

mbartlett21, 2019-07-05:
**Is your feature request related to a problem? Please describe.**
When slicing slopes with only a small deviation from horizontal (<10 degrees), it generates many steps on the slop

**Describe the solution you'd like**
I would like to have skin layers printed with a slope, like in the example below.

**Describe alternatives you've considered**
I did edit the gcode generated by Cura and add some sloped layers on the top, but this would be very tedious

**Affected users and/or printers**
Everyone

**Additional context**
Example results:
[image]

Example GCode with sloped skin (Only tested on my delta printer. This will probable not work on a cartesian printer; I can upload another if you like)
[slopedsurface.gcode.txt](https://github.com/Ultimaker/Cura/files/3362675/slopedsurface.gcode.txt)

Ghostkeeper, 2019-07-08:
This is a big change! In a way, it violates one of the first assumptions that CuraEngine makes when starting a slice, that all layers are horizontal. The case in this ticket is an edge case but the more general problem is a very interesting one that Ultimaker's project managers had put on the long term road map from time to time under the nomer of "non-planar layers". The idea is that the bottom of a print would be flat, but the top of a print would follow the surface topography of the mesh.

The problem lies in the implementation, which makes this a long term project. Following the mesh surface turns many of our current algorithm from 2D into a 3D problem: collision avoidance, generating enough skin layers, calculating overhang, etc.

It'd certainly be interesting to research in this area. However as you said, it's mostly the Delta printers that would benefit from this since the rest can't move the Z position as fast. That makes the target audience for such a feature very limited and the cost-benefit ratio too low for us to work on this right now.

mbartlett21, 2019-07-08:
@Ghostkeeper 
On Onshape, I have been trying an implementation of this.
Currently, I project the layer surface to horizontal and do my infill work on that, then project it back to the original surface. I have found this to work fairly well, with the extrusion amounts being calculated relative to the original surface.

Here is an example GCode for my AnyCubic Delta machine. (z move up to 20mm was added manually to stop head hitting the print)
[Test.gcode.txt](https://github.com/Ultimaker/Cura/files/3367052/Test.gcode.txt)

Ghostkeeper, 2019-07-08:
> Currently, I project the layer surface to horizontal and do my infill work on that, then project it back to the original surface.

Yeah so that would cause the nozzle to crash into the surface if you have multiple pieces and it needs to travel past a raised piece of the print. It would also cause the skin to be too thick in some places and too thin in others.

mbartlett21, 2019-07-08:
@Ghostkeeper 
Currently I have 3 layers that follow the topology of the mesh, then go to horizontal layers after that.
I have assumed that all the horizontal layers are infill and the top two layers that follow the topology are skin

nubnubbud, 2019-07-16:
hmm... This is an interesting idea, though there's the inherent issue of crashing. However, this could just about solve the issue of terrible stepping artifacts on parts with shallow slopes!

I think, however, it would require some way to model the print head in order to avoid moces that would crash into a part, and an unintended consequence is that if a part gets too close to a heat block, chaos may ensue.

my first instinct would be to use the "overhang algorithm", but in reverse, to target shallow, near-horizontal faces, then use a frustum the height of the nozzle to both set the collision mesh and number of top-facing layers to target which ones are viable in a horizontal direction. some considerations:

-will you horizontally expand the layers to calculate which will be too close for the print head to reach

-if you want to get fancy, you could, for example, break it up every 5mm or so to target an entire model. this would be "stepped slanting" or something

-if you want to get really, really fancy, you could make it wait till the next layer would come into contact with the nozzle, then print slanted layers on all the layers currently due for them. this would be "adaptive slanting" or something

-the slanted areas will need their height removed, to produce the proper stepped layers, then the slanted layers build on top of that. How that will work in concave areas I do not know.

kyjak, 2019-08-23:
It is called non planar printing and was added to Slicer, read https://www.reddit.com/r/3Dprinting/comments/ctxns4/nonplanar_3d_printing_functionality_added_to/
Anyway, I vote for this feature! It is GREAT!

thesec, 2019-08-26:
+1 for reopening and migration to Cura

devonjones, 2019-08-28:
+1 for reopening

AlbeRCLife, 2019-09-02:
+1 for reopening

Ghostkeeper, 2019-09-03:
What would you like to have re-opened then? This ticket is already open.

kyjak, 2019-09-03:
> 
> 
> What would you like to have re-opened then? This ticket is already open.

Its probably confusion of the people with some other items. As far as this item includes non-planar printing upgrade, it is great ;)

adrianmihalko, 2019-09-03:
> However as you said, it's mostly the Delta printers that would benefit from this since the rest can't move the Z position as fast

Wait, what? On Ender: https://www.youtube.com/watch?v=gmePlcU0TRw

kyjak, 2019-09-03:
Depends on how fast you expect to print. Normally the top layer is slow anyway to get a nice looking finish.

alexander94dmitriev, 2019-09-06:
That will be a game changes for 3d printing. But I wonder if this feature request should be opened in CuraEngine instead?

Ghostkeeper, 2019-09-06:
We generally prefer to group them here in the Cura repository instead. More people reading it, you see.

tom-truc, 2019-09-13:
This is an amazing features ,+1 to implement it in the next versions.

LostKatana, 2019-11-30:
Using Slic3r there is an experimental version out there:
https://github.com/Zip-o-mat/Slic3r/tree/nonplanar

dxgldotorg, 2020-04-25:
I'd love to see nonplanar mode in Cura, especially since my experience with Slic3r's output has been less than stellar.

LordMelkor, 2020-04-27:
Also experimented with this feature on Zip-o-mat's Sli3r fork. Would love to see something similar in Cura. It was mentioned that this used to be on the Cura roadmap... is that still the case?

cconde, 2020-04-27:
I'd like to see it in Cura, too

EternityForest, 2020-07-05:
It might not do anything for print strength or quality of curves, but what about just randomly varying layer height at different places in the same layer?  An algorithms could be used so that no point was ever more than half a mm below the plane where the print is "supposed" to be, but the layer thicknessses would be random and uneven, eliminating the obvious layer lines look.

You'd just have to randomly choose regions to make thinner, then eventually start making them thicker again to bring them up to the proper place.

nubnubbud, 2020-07-06:
> It might not do anything for print strength or quality of curves, but what about just randomly varying layer height at different places in the same layer? An algorithms could be used so that no [...]

actually, that's a good point. if you could create a regular zigzag or 3D ripple on the z-axis that each layer would follow, you could essentially remove layer lines from being a break point in shear forces. tensile strength may weaken a bit, but it would give each layer more surface area to resist bend-breaks and shearing along each layer. 

what you're talking about can be pretty well done already if you're willing to use fuzzy skin. The issue with varying flow and nozzle height like that is that height changes are the slowest movement most machines make, and anything but the best calibration would produce blobs and zits if you did that, because of the minute lag the print head receives from mechanical stresses. However, a very small perlin displacement field for the height of extrusion would do exactly what you're talking about. 

the issue is that cura is primarily not a 3D slicer. it slices things in 2 dimensions then calculates how to fill them, because of the libraries it uses, so the displacement field would require you to put line segments at regular intervals so there's enough points (line ends) for the perlin field to displace and leave the smallest gaps possible. file sizes would reach gigs pretty fast even with my interpretation, I think. That also increases firmware lag in some cases, if it's too dense.

nesdnuma, 2020-07-10:
+1 for non-planar printing!

Bob90, 2021-02-10:
> It might not do anything for print strength or quality of curves, but what about just randomly varying layer height at different places in the same layer? An algorithms could be used so that no [...]

Someone tried this https://hackaday.com/2016/07/27/3d-printering-non-planar-layer-fdm/

cfontes, 2022-06-08:
bump here!

probonopd, 2022-09-17:
Wouldn't it nice if one could print nonplanar curved top surfaces? These are often used in industrial design, e.g., here:

[image]

I imagine that most of the layers would be printed normally, but then a few nonplanar ones on top, and finally nonplanar ironing. Something along these lines (section view of the layers):

[image]

maxlinux2000, 2022-09-17:
+1
wait!!
+2!! :-)

petterreinholdtsen, 2023-01-17:
Just came across  https://yewtu.be/watch?v=1i-1TEdByZY marketing non-planar 3D printing.