Ultimaker/Cura #5981: Preview line width is very big.

mbartlett21, 2019-07-05:
**Application version**
4.1.0

**Platform**
Windows 10 1903

**Printer**
deltabot

**Reproduction steps**
Load the gcode attached

**Actual results**
The preview shows the line widths (except for the brim) as twice what they should be.

**Expected results**
The preview shows the line widths like it shows the brim

**Additional information**
[TiltedTest.gcode.txt](https://github.com/Ultimaker/Cura/files/3363066/TiltedTest.gcode.txt)

[image]

Ghostkeeper, 2019-07-08:
I think it's misunderstanding what your layer height is. Understandable, since this particular print does a lot of weird things with the layer height!

mbartlett21, 2019-07-08:
@Ghostkeeper 
If i specify the layer height in a comment in the gcode like `;LAYERHEIGHT:0.3`, will it work then?

Ghostkeeper, 2019-07-08:
Well, adding `;LAYER:##` for every layer boundary might fix some things, but I don't think it'll fix this. It looks for changes in the Z coordinate to determine the current layer height. Then it calculates the line width by dividing the extruded volume by the line length and layer height.

Alternatively, perhaps your printer's filament diameter is wrong? If it's 2.85mm instead of 1.75mm, you'd get lines that are 266% as wide.

diegopradogesto, 2019-07-17:
Yes, I think this is the case. If the gcode was generated for a filament diameter 2.85mm, but you are loading it in a Cura printer with 1.75mm filament, the GCodeReader will use this value to calculate the line width. 

Try to load it in a printer with 2.85mm or change the diameter in Machine Settings. If your real printer is set up to print with 1.75mm, then this GCode probably won't work correctly.

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

mbartlett21, 2024-11-29:
I'm happy to classify this as fixed. The line widths on that original gcode now show fairly ok:
[image]

The height of the lines is around half of what the print actually results in, but the width is fine.

closed 2024-11-29 by mbartlett21