Ultimaker/Cura #5982: Multible models on Cura

Exodux1, 2019-07-05:
Ever since I switched to PC from Mac, I've be unable to print more than one model at a time on Cura. When I try, it opens up a new Cura page.

Is this a PC issue or is Mac just more capable using Cura.

Thanks

Ghostkeeper, 2019-07-08:
MacOS by default doesn't allow more than one instance of every application. The expected behaviour is to open a new instance of Cura every time that you open a mesh file.

Perhaps we can change the Windows installer to register the file extension with the `--single-instance` parameter so that Cura will add it to an existing instance if Cura was already open.

LipuFei, 2019-07-10:
@Exodux1  Cura on Windows doesn't run as a single instance by default. To make "Open file with Cura" using an existing Cura instance, unfortunately, we will have to make Cura run as a single instance on Windows by default. We discussed this, and because there are users who prefer using multiple Cura instances on Windows, so we decided not to do this. Thanks for reporting.

closed 2019-07-10 by LipuFei