Ultimaker/Cura #5983: Cura 4.1 adds double G92 E0 command after start gcode.

pinchies, 2019-07-06:
In Cura 4.1 a new bug is added. After the start gcode section, there is normally a G92 E0 command. This command is doubled in Cura 4.1, where as in Cura 4.0 there is only one command.

Example snippet: 

`;FLAVOR:Marlin
;TIME:6224
;Filament used: 4.71714m
;Layer height: 0.15
;MINX:76.326
;MINY:86.434
;MINZ:0.2
;MAXX:147.732
;MAXY:133.495
;MAXZ:47.9
;Generated with Cura_SteamEngine 4.1.0-BETA
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
; -- START GCODE --
G21                     ;set units to millimetres
G90                     ;set to absolute positioning
M106 S0                 ;set fan speed to zero (turned off)
G28                     ;home all axis
M420 S1                 ;turn on mesh bed levelling if enabled in firmware
G92 E0                  ;zero the extruded length
G1 Z1 F1000             ;move up slightly
G1 X60.0 Z0 E9.0 F1000.0;intro line
G1 X100.0 E21.5 F1000.0 ;continue line
G92 E0                  ;zero the extruded length again
; -- end of START GCODE --
G92 E0
G92 E0
G1 F3000 E-5
;LAYER_COUNT:319
;LAYER:0
M107
G0 F3600 X79.721 Y97.782 Z0.2
;TYPE:SKIRT
G1 F3000 E0
G1 F1800 X80.295 Y97.351 E0.02387
G1 X80.673 Y97.093 E0.0391
G1 X81.347 Y96.681 E0.06`

Ghostkeeper, 2019-07-08:
We discussed this issue with the team but don't think it's enough of a problem to fix. Sorry. If you ever figure out what made this happen, let us know.

closed 2019-07-08 by Ghostkeeper

pinchies, 2019-07-08:
I have people complaining to me left right and center that Cura 4.1 broke their printer, while 4.0 works fine. Aside from the z max speed bug, if I were your team I would try having your ear a little closer to the ground - because this is not the first time that a new release has broken things for many many people.

Ghostkeeper, 2019-07-08:
Resetting the E to 0 three times instead of two broke their printer?

We're doing our best to keep our ears close to the ground! I'm spending multiple hours every day processing bug reports, investigating what's wrong with them, seeing if I can fix them or otherwise plan something in. We're honestly doing our best to help out with all of the different printer types out there, but it's a team of 4 programmers serving a user base of millions. On top of that, Ultimaker is paying us to develop for them, which does not include time to develop for other printers. We don't even have the other printers to test with. That's why we have to rely on external contributions to fix those things for them.

We know about the max Z speed bug and fixed that as soon as we could (we didn't hear about it during the beta). But we have to prioritise which things we will fix and which things we won't. I don't think it's fair that you flame me for deciding not to fix a detail like this :disappointed:

pinchies, 2019-07-08:
I know this is not the place to rant. I'm really sorry if this comes across as lacking appreciation - please understand I have the greatest admiration for what your team has created.

I don't know yet what is breaking things for people, I think the z bug was the main one. 

However, in my mind producing good clean output gcode is really an essential goal to avoid all kinds of edge case that crop up from obscure firmware bugs. Cura is an amazing piece of software, but it has (IMHO) gone through too many complete revolutions and not enough time spent on the evolutions and bug fixing.

pinchies, 2019-07-08:
Please know I'm doing my best to help report bugs as I find them too for the range of printers I have contact with, and I will respect your decision if you think this is not a priority to fix.

Ghostkeeper, 2019-07-08:
Sorry, I shouldn't have been that hard on you. I know that the oversight with the maximum Z seam has forced a large portion of our user base to use a workaround to get their printers to work again. But this bug just seems so minor to me that I can't justify spending time on figuring out where it comes from.

The major regressions are almost always happening in something that we can't test: Third-party printers. The beta testers that Cura attracts are largely Ultimaker users since they read the Ultimaker forum more where we post the beta.

On a related side note, we're looking for users with third-party printers to do a targeted beta test for print quality on their printer. If you're willing to do that, we'll send you 3 models (that cover a wide variety of printing applications) to print during every beta release so that you can give us a heads up if the print quality deteriorated for your printer. Completely non-binding of course, all voluntary. Is that something you want to help with? If so, shoot me an e-mail at r.dulek at ultimaker d.t com.

pinchies, 2019-07-08:
Sounds excellent. I'll send through an email.