Ultimaker/Cura #5984: [4.1.0] Ignored options: "outer before inner walls" and "infill before walls"

b4545, 2019-07-06:
**Application version**

4.1.0

**Platform**

Windows 10 v1809 x64
Radeon 19.4.1

**Printer**

Monoprice Maker Select

**Reproduction steps**

Open model, select "infill before walls".
Open model, deselect "outer before inner walls".

**Actual results**

Infill is done after walls. Outer wall is done before inner walls.

**Expected results**

Infill should be done before walls. Outer wall should be done after inner walls.

**Additional information**

Recently I've been playing with these settings: outer brim, inner brim, support brim, support roof and floor, support interface. I'm playing around to see if I can revert a setting which restores the two ignored options. I can provide the curaprofile file if requested.

[image]
[image]

b4545, 2019-07-06:
[FlexExtruder.3mf.zip](https://github.com/Ultimaker/Cura/files/3364967/FlexExtruder.3mf.zip)

smartavionics, 2019-07-06:
Hello @b4545.

On the first layer, if you have brim enabled, it will always print the outer wall first even if the outer wall first option is not enabled. Just try using a skirt instead of a brim and you will see that it prints the inner walls first.

On the first layer, that's not infill that's being printed between the walls, it's skin. Infill is the stuff that gets printed between the skins. So, you will see that when you go to layer 5 it does print the infill before the walls as you would expect.

Hope this helps.

b4545, 2019-07-06:
Thank you for explaining, I understand now. What I'm really trying to achieve on this particular print is an inner brim on the two circles in the first layer. Unless I print really slow those walls don't adhere nicely. Any idea what might be preventing that inner brim?

closed 2019-07-06 by b4545

smartavionics, 2019-07-06:
Perhaps the holes are too small so it won't put any brim in there.

If you want to try one of my Cura builds (Linux and Windows only, I'm afraid) they have some extra features including settings for slowing down the printing of small diameter features:

[image]

You can find my releases at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

They can be installed alongside the Ultimaker releases without conflicting.

b4545, 2019-07-06:
Thank you, I'm going to try it.

natestone, 2020-04-22:
I am also experiencing this issue, but do not have brim enabled.  I do have "skirt" enabled. 

"infill before walls" is checked
"outer before inner walls" is unchecked

However, outer wall is printed first and infill is printed after the walls (resulting in poor overhang performance) during the initial 4 layers.

So, I'm not seeing the expected behavior described by @smartavionics above.  Skirt seems to generate the same incorrect performance.

Is this a bug?  v4.6.0, Mac OSX 10.15.4, Radeon Mac Driver

Even in the "preview" the outer wall is created first as shown in the screen shot below:

[image]

After the "bottom" layers, the infill is done first which matches the setting.  During the bottom layers (4 layers in my case), these settings are ignored.  

Is this the expected behavior?  If so, is there a way to override this behavior?

If it is a bug, should I file a separate report?

Thanks,
--Nate

Ghostkeeper, 2020-04-28:
You have multiple questions there. I'll try to answer them.

> However, outer wall is printed first and infill is printed after the walls (resulting in poor overhang performance) during the initial 4 layers.

I'm unable to reproduce this! It seems to correctly print the inner wall first.
[image]
Maybe a project file could help us deduce the problem here.

> After the "bottom" layers, the infill is done first which matches the setting. During the bottom layers (4 layers in my case), these settings are ignored.

That is expected. Infill is not the same as skin (or bottom, which is part of skin). There is no way to reorder the skin relative to the walls.

> If it is a bug, should I file a separate report?

Yes. So far I'm seeing one thing that would be a bug but I can't reproduce it, and one thing that is not a bug though.

natestone, 2020-04-28:
Thanks for the reply.  It seems it is working as expected if the "skin" portion in not changeable.  I was expecting the ordering to take effect immediately upon start of the print (during the skin phase).  

The "bottom" layers are important in my design as the curvature starts immediately.  Since I cannot change the order on these, layers 2-4 (of the bottom skin) drop overhangs to the ground resulting in the first 10 layers or so total being of poor quality even though layers 5-10 are in the proper order.  Since the first four are in an improper order they form a poor base for the later layers.

So, there is no bug.  The order commands are followed after the skin.  Unfortunately, that is too late for my project.  

Could I put in a feature request to allow re-ordering the skin portion of the print?  My print is basically flawless everywhere else.  Adding support just makes things messy, so I'm not interested in that.  

Thanks for testing.

Ghostkeeper, 2020-04-29:
> Could I put in a feature request to allow re-ordering the skin portion of the print?

No, refer to https://github.com/Ultimaker/Cura/issues/6734 instead. You're not the only person that has had this issue it seems!

natestone, 2020-04-30:
Thanks a bunch for the follow up.  I look forward to when the feature makes it to production.

RyushoYosei, 2021-03-24:
I have been having Cura do this to me recently too.