Ultimaker/Cura #5985: Request: Move head out of print on dual extrusion while waiting for head to heat up

dlc60, 2019-07-07:
Currently CURA (using 4.1) will just stop the head where it is to wait for a toolchange heat-up.  This leaves a blob under the new tool nozzle which creates a bit of a "wall" in a poor location.

Can Settings menu item be added to move the head elsewhere during this heat-up period?
Lacking that, can the Extruder Start scripts be executed before the toolchange heat-up is done?
Or can the Extruder End scripts activate before the toolchange heat-up begins?

I have tried to do this with CURA in both the Extruder Start and Extruder End scripts, and the new tool heat up occurs before the End of the current nozzle and before the beginning of the new nozzle.  There does not appear to be a way to move the head before the firmware stalls waiting for the new temperature.

**Application Version**
CURA 4.1

**Platform**
Mac OSX

**Printer**
Tevo Tarantula with E3D Chimera head running Marlin 1.8

**Steps to Reproduce**
Configure a dual extrusion printer.
Set Extruder Start g-code: G1 X15 Y90 F4000
or 
Set Extruder End g-code: G1 X15 Y 90 F4000
Load a model
Assign extruders to each model half
Merge models
Slice, Print via Octoprint

**Actual Results**
Print head stops over the print until new head is up to temp, then prints, then moves to the G1 position.
Then moves right back, then sets the active head temp to 175 (I have inactive set to 165)
Then stops over the print
Then moves to defined location and heats BOTH heads up to the active temperature.
Then moves to the print and turns down the inactive head.
"Rinse repeat"

**Expected results**
When toolchange occurs:
Head moves to set X/Y location and stops
inactive head should set M104 S165...
active head should set M109 S190
Until active head reaches 190 then
Head should move to the print and start printing

The above behavior was observed when the G1 X15 Y90 F4000 g-code was located in the Extruder Start script.
I am pretty sure that this was the sequence that I observed.  The temperature settings were pretty bizarre, I don't even have a 175 setting in my configuration!

[CFFFP_Phil_Color_2.zip](https://github.com/Ultimaker/Cura/files/3365356/CFFFP_Phil_Color_2.zip)

dlc60, 2019-07-07:
I forgot to add this essential bit.
When I add the G1 X15 Y90 F4000 into the Extruder End script, the printer does what is expected.  It moves to the assigned location and waits for the head to heat up. 

DLC

dlc60, 2019-07-07:
Sorry, no, it doesn't work in 3.6 either.  I thought that it did, but only once.  Then I started getting the inactive head set to 180, or both heads set to 190.  Sometimes the head would wait at the set point, and sometimes it would go there and then go right back.  I get a MUCH better print because at least the head does not stall above the print while waiting for the temperature to change, but it isn't the behavior I expected to see.

DLC

Ghostkeeper, 2019-07-08:
Yes, this already exists. It's in the following settings:

* machine_extruder_end_pos_x, machine_extruder_end_pos_y: Where the nozzle needs to finish, just before switching extruders.
* machine_extruder_start_pos_x, machine_extruder_start_pos_y: Where the nozzle starts, after switching extruders. Cura never actually outputs these coordinates but they are necessary to get accurate print time estimates if the act of switching extruders leaves the nozzle in a different position, such as with the Ultimaker 3 (which is why we built it).
* machine_extruder_end_pos_abs, machine_extruder_start_pos_abs: Whether the start and end positions are absolute or relative to the current position.

By default, these positions are relative and set to [0,0], which prevents the nozzle from moving to weird locations for unknown printers. If a specific location to switch extruders is desired, set the X/Y coordinates properly and make the start/end positions absolute.

I'm quite certain that this works correctly because I'm using it every day to print stuff on the Ultimaker printers.

closed 2019-07-08 by Ghostkeeper

fieldOfView, 2019-07-08:
> Those are machine settings and so they are only available when you modify the configuration files.

... or when you install the Printer Settings plugin from the Marketplace.

Ghostkeeper, 2019-07-08:
I don't think that the Printer Settings plug-in would show those settings since they are in fdmextruder.def.json not in fdmprinter.def.json.

dlc60, 2019-07-08:
OK, so the answer is "no", there is no way to use the CURA UI to do this.

Thanks anyways,
DLC

tphan26363, 2020-08-11:
I don't know it is possible but please re-open this issue.
 I think the problem is not Cura fault but I think cura can solve this problem easily.  Here is the fact that  I have found:
-After a T0 or T1 change head command, the head move and rest right above the part.  If the nozzle was set to recover the setting temperature, the head will wait right there and deposit blob on to the printed part.
-The problem is not Cura but it is part of the command  to change head.  After executed a head change, the firmware will try to return back to the position it was before the change which is right above the part. 
-The execution of the T(i) command is not wrong neither.  It was trying to keep track of it location which is return to where it was before change head
-My suggestion is that Cura can insert a wait position BEFORE instead AFTER  command  T(i) executed.
So when the T(i) command completed it will return to where is was before which is the rest position defined by user
-I have successfully doing this by post processing the gcode file.  It probably an easy fix for CURA

fieldOfView, 2020-08-12:
Couldn't you add a simple `G0 X10 X10` to the Extruder end gcode snippets of both extruders in the machine settings?

tphan26363, 2020-08-13:
Yeah.  I can do that but it seem like it doing extra weird movement. i just want to consolidate the move.  no extra weird movement

Ghostkeeper, 2020-08-13:
> -My suggestion is that Cura can insert a wait position BEFORE instead AFTER command T(i) executed.

This is exactly what the `machine_extruder_end_pos_x`/`y` and `machine_extruder_start_pos_x`/`y` are. If you're writing a machine definition, please adjust those. If you are not, please ask the one who wrote it to make an improvement, or change them yourself (e.g. using the Printer Settings plug-in from the Marketplace).

tphan26363, 2020-10-01:
OK after suffer for a long time I found out what is the problem.  I think Cura creates unnecessary movement after change tool.
  As see below after the the tool change.  Cura some how create a move to the last position of tool 0 which is not really needed.  Since it will blemish the T1 color on to T0 position( usually at the seam position).  This is ridiculous.  Only Z position need to remember,  The T1 should start at the position that need to be printed for T1.  This happens every time TOOL CHANGE HAPPEN.  IT DOES NOT MATTER IF WE PUT SCRIPTS TO THE BEFORE OR AFTER TOOL CHANGE
**G1 X208.143 Y202**  //T0 end here
G1 F1500 E20.47695
G1 F600 Z1.4
G92 E0
G0 X195 Y100 F12000
T1
G92 E0
M105
M109 S205
G1 F600 E21.7
M204 S5000
;MESH:wall2.stl
G0 F5400 X208.143 Y202 Z1.4 // T1 return here which is not needed since this is the last position of the T0.  Only z movement needed

Ghostkeeper, 2020-10-02:
That is not a Z movement. It was already at `Z1.4` due to the third line of your snippet. That is a movement to the starting position of the next layer, I'm guessing.

tphan26363, 2020-10-02:
I think so too But the XY position is the last position on Tool-0,  The Tool1 should not need to move there.  It just drop oozed material to tool 0 printed area.  G0 X195 Y100 F12000 is inserted by me in the :After tool change" with the hope that it will move there instead.  But it is not happening.

tphan26363, 2021-01-18:
This is ridiculous way to arrange a switching tool head.  I totally disable the switching tool head retraction from cura  and change marlin to have it own strategies to switch tool head and retraction.  Come out with much cleaner print.  Thanks for nothing Cura.

Ghostkeeper, 2021-01-25:
Luckily it's all optional. Cura can just be configured such that a tool change is a single `T1` command if your printer knows it better.