Ultimaker/Cura #5986: Z-Hop slows down Printer MASSIVELY

HACKER-3000, 2019-07-07:
**Application version**

Cura 4.1

**Platform**

Windows 10

**Printer**

Creality Ender 3

**Reproduction steps**

Turn on Z-Hop

**Actual results**

Printer randomly slowed down MASSIVLY tot he Point where you cant really see it moving!

**Expected results**

Z-Hop Works just fine.

**Additional information**

[A reddit post about this](https://www.reddit.com/r/ender3/comments/c6lsmo/wierd_thing_with_my_e3/)

mbartlett21, 2019-07-07:
@HACKER-3000 
Can you try setting a z max speed?

arminbl, 2019-07-07:
Same here. 
Cura inserts a very big/random feedrate into the G-code. when activating Z-Hop 
It looks like this snipet below:

G1 F17987547480000 Z1.3
;MESH:test.stl
G0 F3600 X213.993 Y115.598 Z1.3
;TYPE:WALL-INNER
G1 F17987547480000 Z0.3

mbartlett21, 2019-07-08:
This seems to be a duplicate of #5753 , which is already fixed and will hopefully be fixed in the next release of Cura.
As a workaround, you can set a z max speed that is not 0 in Cura

Ghostkeeper, 2019-07-08:
Yup, duplicate of #5753.

closed 2019-07-08 by Ghostkeeper