Ultimaker/Cura #5988: [4.1.0] Cura crashes the whole PC (Windows 10)

Skybeat, 2019-07-07:
**Application version**
4.1.0
**Platform**
Windows 10 Version 1903 Build 18362.175
CPU: AMD Ryzen 7 2700X
GPU: Nvidia Geforce GTX 1060 6GB

**Printer**
Creality Ender 3

**Reproduction steps**
Start Cura and wait

**Actual results**
Cura crashes the whole PC extremly hjard. the only thing i can do after that is: remove the power plug from power adapter of the PC to turn off the pc. There is not other way to turn off the pc after Cura crashes the PC.
Withhin the crash my 2 monitors gose black but the PC is still on but not useable. It is also not poisble to press and hold the power butoon of the PC to turn him off!

**Expected results**
the PC crashes very hard and cant turn off the normal way.

**Additional information**
I have this issue with every other cura version since i start with cura v. 3.5.0.
Some times Cura works for 10 mionutes, some times only for one minute.
[stderr.log](https://github.com/Ultimaker/Cura/files/3365618/stderr.log)

smartavionics, 2019-07-07:
See https://github.com/Ultimaker/Cura/issues/5265.

hagre, 2019-07-07:
Maybe the following is related to your Bugreport.

Since some weeks i had a crash of cura on ever start (Screen/WIndow maintains white and blocks the rest). Today i found a workaround for Cura 4.0 and 4.1 on my Windows 10 1903 Notebook:
I have a **Intel UHD 630** and **NVIDIA GTX 1060 6GB**, if i select the NVIDIA ->crash, No problem on the UHD 630 as long as i keep the window on the notbookscreen. If i move the Window on the second /external screen, the NVIDIA GPU comes into play and after 1 or 2 clicks (right mouse button) -> crash again.
So i just can use cura on my primary display with Intel UHD 630 selected (otherwise it crashes with a white screen during start)

(Driver version NVIDIA 25.21.14.1967 dated 17.03.2019)

Maybe that is also a workarond for you.

Skybeat, 2019-07-07:
> Maybe the following is related to your Bugreport.
>
> Since some weeks i had a crash of cura on ever start (Screen/WIndow maintains white and blocks the rest). Today i found a workaround for Cura 4.0 [...]

Thank you for the tip but it want work for me since i work on a desktop pc and have only the Nvidia grafic card.
Until now i think also that this is a issue with Cura and my Nvidia card. i have also installed different drivers for my card but the issue is still happen

Ghostkeeper, 2019-07-08:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/5265. Please continue the conversation there, to keep it all in one place.

closed 2019-07-08 by Ghostkeeper