Ultimaker/Cura #5989: Long running prints USB timeout

obiben, 2019-07-08:
**Application version**

4.1.0

**Platform**

Windows 10 home x64
Version 1809
OS build: 17763.557

**Printer**

Monoprice mini delta

**Reproduction steps**

Printing long (>2hour) prints, there seems to be an increasingly high probability that the print will stop midway. It seems like windows is turning off the usb port used to save power. 

I don't think I've completed a single print over 5 hours via USB.

**Actual results**

The print will start normally
After a couple hours, the printer will stop moving. It will stay at temperature but will not move again until cancelling the print via the on-printer interface (cannot cancel the print through cura)

**Expected results**

A running print should never allow windows to shut off the USB connection

**Additional information**

Ghostkeeper, 2019-07-09:
We haven't encountered this ourselves yet, but it's been a long time since I've printed something bigger than 2cm via USB cable. Perhaps this is something that a Windows update destroyed for us?

obiben, 2019-07-09:
Here's a log of a test I did tonight with selective USB suspend off. This morning at 6 I was giddy it was just that, then...

2019-07-08 20:52:53,605 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.addCallback [28]: on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x0000016884275798>>] added.
2019-07-08 20:52:53,605 - DEBUG - [Thread-13] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started.
2019-07-08 20:52:55,239 - DEBUG - [Thread-13] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM3 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
2019-07-08 20:52:56,768 - DEBUG - [Thread-13] USBPrinting.AutoDetectBaudJob.run [80]: Detected baud rate 115200 on serial COM3 on retry 0 with after 0.00 seconds.
2019-07-08 20:53:18,568 - DEBUG - [MainThread] UM.Controller.setActiveStage [149]: Setting active stage to PreviewStage
2019-07-08 20:53:18,580 - DEBUG - [MainThread] UM.Controller.setActiveView [91]: Setting active view to SimulationView
2019-07-08 20:53:18,664 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/ActionButton.qml:70:9: QML Label: Binding loop detected for property "width"
2019-07-08 20:53:18,666 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/ActionButton.qml:70:9: QML Label: Binding loop detected for property "width"
2019-07-08 20:53:18,673 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/ActionButton.qml:70:9: QML Label: Binding loop detected for property "width"
2019-07-08 20:53:18,676 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/PreviewStage/PreviewMain.qml:22:5: QML ActionPanelWidget: Binding loop detected for property "width"
2019-07-08 20:53:18,730 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:18: TypeError: Cannot read property of null
2019-07-08 20:53:18,732 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:20: TypeError: Cannot read property of null
2019-07-08 20:53:18,735 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:21: TypeError: Cannot read property of null
2019-07-08 20:53:18,737 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:42: TypeError: Cannot read property of null
2019-07-08 20:53:18,739 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:43: TypeError: Cannot read property of null
2019-07-08 20:53:18,742 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:44: TypeError: Cannot read property of null
2019-07-08 20:53:18,744 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrintSetupTooltip.qml:67: TypeError: Cannot read property of null
2019-07-08 20:53:18,746 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrintSetupTooltip.qml:69: TypeError: Cannot read property of null
2019-07-08 20:53:18,749 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrintSetupTooltip.qml:71: TypeError: Cannot read property of null
2019-07-08 20:53:18,751 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:36: TypeError: Cannot read property of null
2019-07-08 20:53:18,753 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:37: TypeError: Cannot read property of null
2019-07-08 20:53:18,755 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:49: TypeError: Cannot read property of null
2019-07-08 20:53:18,758 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:59: TypeError: Cannot read property of null
2019-07-08 20:53:18,760 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:61: TypeError: Cannot read property of null
2019-07-08 20:53:18,762 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:63: TypeError: Cannot read property of null
[... cut ...]
2019-07-09 07:23:06,222 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/MonitorButton.qml:22: TypeError: Cannot read property 'activePrinter' of undefined
2019-07-09 07:23:06,224 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/MonitorButton.qml:21: TypeError: Cannot read property 'acceptsCommands' of undefined
2019-07-09 07:36:08,768 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles

Ghostkeeper, 2019-07-10:
So for some reason the device doesn't give a response any more after a night of printing.

We're not able to reproduce this on our Windows computers, but we haven't been able to try for very long. We've decided that we'll wait to hear whether other people are encountering this as well or whether it is something isolated to some configuration option on your computer.

obiben, 2019-07-10:
That's fair. 

Apparently most people just regard printing through USB on windows as unstable and either set up octoprint or use a SD card. That's most of the responses I've gotten from the Monoprice mini delta facebook group.

My initial thought when reporting this was that the CURA software could do something with regards to USB drivers to prevent selective suspend, but since it freezes up with selective suspend turned off, I'm as confused as you are.

Ghostkeeper, 2019-07-11:
Something probably triggers it, but working with Windows in this sort of thing is a nightmare.

Ghostkeeper, 2021-01-11:
In #9054 people have been continuing this discussion. Since that issue now has more analysis and possible workarounds than this one, I'll close this as a duplicate (even though it was earlier).

closed 2024-11-22 by obiben