Ultimaker/Cura #5991: Interface unusably slow

fvlasie, 2019-07-08:
Hello,

I am noticing incredible slowness when trying to use the Cura interface. It literally take several seconds to respond to a button push. Trying to rotate the model takes several mouse clicks just to get the model to select. I am not using a slow computer either: Mac Pro 3.46Ghz 6 core, 32GB RAM, 1TB SSD, Mac OS 10.13.5.

Thank you

FV
[slowcura.mp4.zip](https://github.com/Ultimaker/Cura/files/3369804/slowcura.mp4.zip)

Ghostkeeper, 2019-07-09:
The slowness when enabling support is known. There's a lot of settings that need to be updated when this happens. Some settings have it worse than others, but the "Enable" checkboxes are particularly bad in this aspect.

The other issue is not that it's slow, but that there are sometimes zones on the screen that you can't seem to click on. Also a known issue, but hard to reproduce if you actually want to reproduce it: https://github.com/Ultimaker/Cura/issues/5880

closed 2019-07-09 by Ghostkeeper

fvlasie, 2019-07-09:
OK! thanks for the info.