Ultimaker/Cura #5994: buf is 0

amparodiazvera, 2019-07-09:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

smartavionics, 2019-07-09:
Hello @amparodiazvera , I think you're going to have to provide more information than that if there is any chance that the issue will be investigated. How about some details of what went wrong?

amparodiazvera, 2019-07-09:
Hi! I have an Anet A8. And cura 3.6. Every time i try to print with the SD card, the print starts, butthe “buf” on the lcd screen starts from 16 to 0 and the movements of the printer starts to stop. The quality prints are turning horrible because of this. 
I later install the cura 14.07 and printed the same stl file and everything goes great!
I tried with cura 4.1.0 and the same, i tried with another sd car and the same. 
It appears to be a problem with saving de gcode!
Do you know how can i fix this???

Amparo Diaz

Ghostkeeper, 2019-07-09:
I edited the original post. The stuff between the \

Ghostkeeper, 2019-07-10:
I'm going to assume that was the problem. If that doesn't fix it, let us know and we'll open the ticket back up to see if there's anything Cura can do.

closed 2019-07-10 by Ghostkeeper