Ultimaker/Cura #5996: Snap package

mikeroyal, 2019-07-10:
I was wondering if you might have an interest in a [snap](https://snapcraft.io/) version. I would help make the snap and submit a PR if there's any interest.
[Crafting your first snap](https://tutorials.ubuntu.com/tutorial/create-your-first-snap#0)

Ghostkeeper, 2019-07-10:
We've chosen AppImage as our solution instead of this and don't have any incentive to change that right now. Sorry.

closed 2019-07-10 by Ghostkeeper

probonopd, 2019-07-31:
@mikeroyal AppImage is specifically designed so that

* The author of an application is in full control over the whole end-to-end software distribution experience with no gatekeepers nor middle-men services in between
* The user can keep many different versions alongside each other
* The application can be managed without interfering with the operating system's package manager, and can never change the system

@mikeroyal what advantage would you think would Snap give you? Could the same advantage be achieved by improving AppImage?