Ultimaker/Cura #6000: Defacing with 4K NON-HiDPI display

sl1pkn07, 2019-07-11:
**Application version**
GIT 6941af002

**Platform**

Archlinux
Nvidia RTX2060 with latest vulkan drivers

**Printer**

**Reproduction steps**
build from GIT and run it

**Actual results**
[image]

**Expected results**
IDK ,because i can't disable the HiDPI support with the QT environment variables
https://doc.qt.io/qt-5/highdpi.html#high-dpi-support-in-qt or cura settings

**Additional information**

Qt 5.13 from repos (archlinux)
uranium and curaneguine from GIT
dark or light theme doesn't matter

fieldOfView, 2019-07-11:
Please try the AppImage from software.ultimaker.com

sl1pkn07, 2019-07-11:
seems look good

[image]

upscaled, but without defacing

fieldOfView, 2019-07-11:
In that case, this is a duplicate of https://github.com/Ultimaker/Cura/issues/5487

Cura is tested with Qt 5.10, you are using a newer version. Ultimaker only supports the AppImage version for Linux.

Ghostkeeper, 2019-07-12:
Yes, those are bugs that seem to arise from Qt 5.12+. We require 5.10 right now.

closed 2019-07-12 by Ghostkeeper