Ultimaker/Cura #6001: Gradual Infill - Disconnected line segments printed in mid-air

dstulken, 2019-07-11:
**Application version**
Cura 4.1.0 (Mac)

**Platform**
macOS 10.14.5

**Printer**
(doesn't matter)

**Reproduction steps**
* Load a model with inward sloping features on the sides, or pockets on the top surface. 
* Enable a relatively sparse infill, such as Lines at 4%.
* Enable Gradual Infill with a low Step Count (ie:1) and a relatively thick Step Height (ie: 6mm) to better highlight the problem.
* Observe the areas of infill below the inward sloping sides, or top pockets.

**Actual results**
Assume a cube shape with features inset into it.
* Above the transition point, the higher-density infill layers begin and end at walls as expected. 
* For the areas where the transition starts lower due to model features, areas near inward sloping sides or pockets from the top may have infill generated as partial segments positioned under the model elements.  These segments are printed in free-air, with no support underneath or any connections to adjacent infill lines, so the extruded plastic is dragged around to impact or clump up on whatever the nozzle travels to next.  If the Step Height is a large value, or comparatively large compared to the layer height, then many successive layers will attempt to build upon and fan out from these lines, resulting in more and more useless (or detrimental) mid-air printing until the top layers are finally printed.  
* Depending on the model, features printed above this "infill" will fail, as the expected infill beneath them is not actually present.

**Expected results**
* Gradual Infill segments should always begin and end at/above a point where previously extruded plastic exists.  This could be walls or infill of the prior layer, or features of the current layer if the wall/infill print order permits.  I believe this would just be an extension of the line segments that the current Graduated Infill algorithm produces.

**Additional information**
Simplified test case which demonstrates the problem:

Use the following code in OpenSCAD to generate a STL.  
This is a model of a cube with a V-shaped diagonal notch taken out of one top edge.
```
difference()
{
	cube([100,100,50]);
	
	translate([-1,50,15])
	rotate([45,-45,0])
	cube([100,50,50]);
}
```

Slice in Cura 4.1.0 using the following settings:
Layer Height 0.4mm
Line Width 0.6mm
Infill Density: 4%
Infill Pattern: Lines
Gradual Infill Steps: 1
Gradual Infill Step Height: 6mm

Observe the effects near the V-cut on layers between approximately 96-106, where multiple line segments have been constructed upon each other in free air.

I have attached three screenshots demonstrating the results.
Two screenshots are of the test case above, while a third (not described) is depicting a comparable result from cylindrical pockets inset from the top layer. 

[image]

[image]

[image]

Ghostkeeper, 2019-07-12:
Indeed, this is even a problem for *everything* printed on top of sparse infill. We've found that it usually corrects itself after a few layers. That's why we have the Gradual Infill Step Height at 1.5mm and the Top Thickness at 1mm by default.

dstulken, 2019-07-12:
I think you're lucky if it does... Depending on the model, it may *never* correct itself.  I lost a long print to this, which is what prompted my bug report.

Here's a more severe example, the box with cylindrical pockets in the top surface.  I used 2 Gradual Infill Steps and a 5mm step height on this one.

At layer 48, we can see that Gradual Infill has kicked in under the pockets, and has started building up stacks of line segments.  However, 8 of the 16 pockets have their gradual infill lines printed in mid-air, and none of the successive gradual infill lines ever connect to the adjacent infill lines.
[image]

At layer 49, the top layers begin printing.  None of them would be perfect at this point, but with a sufficient number of top layers, we would expect them to recover.  However, again, fully half of them are printed over open air, and the plastic debris is just dragged around to wherever the nozzle goes next.
[image]

Jumping ahead to layer 110, the majority of the model has now been printed.  However, half of the pockets *still* have zero connections to anything solid.  That's a tremendous amount of air printed mess, and a total failure of the desired print.  
[image]

In the end, it wasn't until layer 111 that the next Gradual Infill density change kicked in and finally connected something.  That's a full 75 layers from where the first detached Gradual Infill printing started.  Disable Gradual Infill, and this piece would have been printable...

Ghostkeeper, 2019-07-15:
What I mean to say is that if you disable Gradual Infill and reduce the infill density to a quarter it would not be printable either. You'd not expect it to, of course. But it's exactly the same problem. Gradual infill reduces printing time, but you will have to pay attention for some models so we won't enable it by default.

Aligning infill lines to support whatever is going to be printed above them is a difficult problem, because the problem space is immensely diverse. If you can come up with a way to do that, we're listening with both ears!

dstulken, 2019-07-16:
Completely agree, @Ghostkeeper.  
I wouldn't expect any of the current infill patterns, or the Gradual Infill modifier, to automatically support everything on the top surface.  

But I **would** expect any infill pattern to not print fully disconnected line segments over thin air, or cover a print inside and out with globs of spaghetti or knock over small features as part of its normal operation.   The infill patterns certainly wouldn't do that with Gradual Infill disabled.   That's why I led with the side-notch example - it's a model with modest overhang and no top indentations, that is 100% printable with just about any infill that can support a flat top roof.   You wouldn't expect that have loose plastic rattling around inside after a print, or scars on the walls from collisions.

So just to be 100% clear:  The bug report is for disconnected infill lines being started and ended, sometimes entirely printed, over free space, with nothing they could be connected to or deposited on.   I only included the cylindrical pocket example to demonstrate that this issue isn't one that can be reasonably expected to "resolve itself" after a few layers.    

It's a real deal-breaker for a lot of models, where this infill feature would otherwise be a phenomenal saver of both print time and material.   

Hope that helps...

dstulken, 2019-07-16:
To solve the problem, the most elegant solution would be to take the infill line segments that Gradual Infill currently generates, and extend their endpoints until they connect with something - either another infill line, a wall, or something on the layer beneath.   That would resolve the problem, eliminating the air printing, plastic debris, and dragging plastic damage.  I've attached a picture of what this would look like, with the revised infill line segments in magenta.

[image]

A more crude solution, perhaps easier to code or of less computational demands, would be to just enable the entire infill stripe from wall to wall, any time any part of it is called upon.  I don't know how the current code works, but this could even be simpler than what is in Cura right now, which aligns and truncates these segments.  This method would use a bit more plastic, but would always guarantee a printable part with no loose ends to the infill.  In the big picture, it would still save a significant amount of material over the alternative of not using Gradual Infill - and any plastic that _was_ printed would at least have purpose, so the overall result is still less wasted plastic than the current implementation in Cura 4.1.0.

[image]

Ghostkeeper, 2019-07-17:
> But I **would** expect any infill pattern to not print fully disconnected line segments over thin air,

It's not in thin air. It's over a very sparsely printed pattern and you have the bad luck that it happens to land in the holes between infill lines. The reason why you're saying that it works for normal infill is because there you tend to use higher infill densities than in the lowest layer of gradual support. For instance, here is a case where normal infill goes wrong in exactly the same way:

[image]

I know what you're saying though. You'd want the infill lines in your example to continue to lean on the other infill lines, and you'd want the skin lines in my example to continue to hit the infill.

The high-level algorithm for gradual infill is currently this:
* Take the normal infill area.
* Intersect the top infill layer with the layer 1 gradual step below it (1.5mm) to produce the area that needs to be filled with the first infill reduction. Do this for every layer.
* Intersect the top second-gradual-step layer with the layer 1 gradual step below it again to produce the area that needs to be filled with the second infill reduction. Do this for every layer. And so on for every gradual step.
* Fill every gradual step with the infill pattern of the desired density (halving the density at every step). This uses the normal infill pattern algorithms.

The cinch is in the last step there. We simply fill a certain area with an infill pattern of a certain density. To this end we call upon the existing infill algorithms for normal infill. This highlights a problem with your change request: How should we make it work with all infill patterns? For grid infill it's logical to extend the pattern with straight lines. For cubic infill you'd somehow want to extend it downwards as well. For gyroid you could decide to continue in straight lines, but those would cross often, in random directions depending on where in the curve the line happened to end. For concentric infill there **are no endpoints** to connect to. There are also numerous settings that would affect this. Stuff like "Connect Infill Lines" shift the lines pattern inwards by half a line width and also remove most of the line endpoints to connect to.

Additionally there is the issue of how far a line should continue towards the next infill line. Computing a ray intersection to the next infill line can indeed be costly, so it'd have to be towards the edge of the polygon. Then we have a slight problem of expectation: "Why are there so many lines across my infill when I requested a density of only 5%?"

Ghostkeeper, 2020-01-01:
We've merged a feature that addresses this issue for the next release: https://github.com/Ultimaker/CuraEngine/pull/1030
Should be a lot better if that setting is enabled!

closed 2020-01-01 by Ghostkeeper