Ultimaker/Cura #6003: 4.1 White screen but menus are fine.

SaltyLiquorice, 2019-07-12:
**Application version**
4.1.0

**Platform**
Win10 1803

**Printer**
Both UM3 and SidewinderX1

**Reproduction steps**
Open Cura

**Actual results**
White screen

**Expected results**
Not a white screen

**Additional information**
[image]

Heres the log of the last time I opened Cura:
[BadLog.txt](https://github.com/Ultimaker/Cura/files/3386118/BadLog.txt)

fieldOfView, 2019-07-12:
What GPU does your computer use? Cura is having trouble opening an OpenGL context.
Interestingly, Cura is "trying to" crash with a critical error, but doesn't.

SaltyLiquorice, 2019-07-12:
Dedicated, 1080ti.

fieldOfView, 2019-07-12:
It is unlikely that reinstalling Cura will do anything. Updating drivers may help.

SaltyLiquorice, 2019-07-12:
And a PC restart fixed it.

closed 2019-07-12 by SaltyLiquorice

fxh1357, 2019-07-13:
I got the same issue - Restarting the PC can't be the fix^^

Ghostkeeper, 2019-07-15:
PC restarting might be the fix if you updated your graphics drivers and didn't restart the PC yet afterwards.