Ultimaker/Cura #6005: [4.1.0] Text too high in text boxes in machine settings window

kockahonza, 2019-07-12:
**Application version**

 Cura 4.1.0 (Linux)

**Platform**

Arch linux with i3wm, X running on an integrated intel GPU with i915 as the driver

**Printer**

Rebelix X2, but not really relevant

**Reproduction steps**

Open the machine settings dialog of any printer.

**Actual results**

The window opens but text in all of the single line text input boxes is too high to be readable.

**Expected results**

**Additional information**

[image]

Ghostkeeper, 2019-07-15:
I'm unable to reproduce that. Here's how it looks for me on Ubuntu 19.04 through the AppImage for Cura 4.1:
[image]

Are you using the AppImage version of Cura that we distribute? If not, could you try if the issue remains with [this version](https://github.com/Ultimaker/Cura/releases/download/4.1.0/Ultimaker_Cura-4.1.0.AppImage) with the proper dependencies?

kockahonza, 2019-07-16:
I first tried the arch package but after I discovered the bug I downloaded the same Appimage you linked to. The issue happens with both.

Also here I am attaching the log file in case it's interesting (i deleted the log, then opened cura, navigated to the machine settings window and then closed it).
[cura.log](https://github.com/Ultimaker/Cura/files/3397733/cura.log)

Ghostkeeper, 2019-07-17:
Hmm, well looking at the code there is nothing really special that we're doing in this dialogue, compared to the rest. The width and height of the box are entries from the theme. All the fonts are properly set from the theme as well. Everything seems to be properly using the pixel size scaling for high DPI screens (since everything is from the theme). Only the column spacing in that page is a hard-coded number, but it's properly multiplied by the `screenScaleFactor`. In the log there's a few binding loops but nothing in this dialogue.

Are you using a high DPI screen?

kockahonza, 2019-07-17:
I have a dell XPS 15 9560 with a full HD 15" screen so i don't think so.

I'm also really confused by the issue but so far I didn't find any other place where it happened.

kockahonza, 2019-08-09:
Today I updated to 4.2.1 and the issue is no longer there. Just thought I would let you know. It was probably related to some other problem that got fixed.

Ghostkeeper, 2019-08-13:
We did change some things in the layout there and in the font system in 4.2 for a different issue. Looks like that fixed it for you.

closed 2019-08-13 by Ghostkeeper