Ultimaker/Cura #6006: Cura not fully loading 4.1.0

BenGoodyear11, 2019-07-13:
I've been having issues with the cura application not loading all the way and not rendering all the icons, it works but the model is not rendered and so It's hard to know if my settings have been applied properly. I'm new to printing please ask if any other additional information is needed

[image]

nallath, 2019-07-15:
Could you please fill in the entire bug report? We also need things like log files and data about what machine you are running on (Processor, video card, what version OS, etc)

GOLDIdeluxe, 2019-07-15:
I have exactly the same issue.

**Application version**
Cura 4.1

**Platform**
Win 10 Pro 1903, build 18362.239
GeForce GTX 760 Driverversion 23.21.13.8813 (installed by windows)

**Printer**
Issue occurs without a printer installed, Printer is Creality CR-10S

**Reproduction steps**
After Reinstall, closing the program and re-opening it, the issue is back. Deleting the config doesn't change anything.

**Additional information**
See the screenshot above, it looks exactly the same on my system, logs are attached
[cura.log](https://github.com/Ultimaker/Cura/files/3392148/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/3392149/DxDiag.txt)

Ghostkeeper, 2019-07-16:
In your logs it's indicating that it's crashing here:

https://github.com/Ultimaker/Uranium/blob/c0c23d590c0a6ab218e396c82d6efc271b9f45b3/UM/View/GL/OpenGL.py#L54

`currentContext()` returned `None`, meaning that it failed to create the OpenGL context. This is why the interface renders fine but the 3D scene and images do not. Interestingly, DxDiag.txt does not list the supported OpenGL versions... :( But your GPU is listed to [support OpenGL 4.3](https://www.geforce.com/hardware/desktop-gpus/geforce-gtx-760/specifications) so that should be okay... You could try updating your video drivers, but if that doesn't help I'm not sure what to do about this issue.

GOLDIdeluxe, 2019-07-16:
Thanks for the reply, i updated to driverversion 24.21.13.9924 via windows device manager, but unfortunately it doesn't solve the issue.

GOLDIdeluxe, 2019-07-17:
I completely removed my video driver and reinstalled it, now it works!

Ghostkeeper, 2019-07-17:
Okay, well at least that is a workaround for people that encounter the same issue then.

closed 2019-07-17 by Ghostkeeper