Ultimaker/Cura #6007: [4.1.0] crash when switching printers in Monitor view

rmpel, 2019-07-13:
**Application Version**
4.1.0

**Platform**
MacOS Mojave 10.14.5
Windows 10 PRO (1903)
MacOS High Sierra 10.13.6 
Ubuntu (can't remember version)

**Printers**
Creality Ender 3 and Ender 5 (Ender 5 uses same profile as Ender 3, but with changed build volume)

**Plugins**
OctoPrint Connection 3.5.6

**Steps to Reproduce**
1. Open Cura
2. Click Monitor tab
3. Switch to different printer

**Actual Results**
Cura crashes

**Expected results**
Display video feed from the newly selected printer

**Additional Information**
When switching printers like this:

- (when on Monitor) click "Prepare" tab
- Switch printer
- click "Monitor" tab

I can switch back and forth to my hearts content
When switching **while on the monitor tab** Cura crashes

I use 2 printers
- Ender 3, with an MKS Gen L/TMC2208 mainboard, Marlin 1.1.9, Octoprint (latest version)
- Ender 5, with a Reality Melzi 1.1.5 Silent (TMC2208) mainboard, Marlin 1.1.9, Octoprint (latest version)

**Additional testing is done**
- it does not matter if a model is loaded
- it does not matter which material is chosen
- it does not matter if the application has been running a long time or freshly started
- it does not matter if I reboot my Mac
- it happens on all platforms

Crash report by OSX Mojave 10.14.5:
`Process:               cura [1725]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.1 (4.1.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [1725]
User ID:               501

Date/Time:             2019-07-13 19:00:26.970 +0200
OS Version:            Mac OS X 10.14.5 (18F203)
Report Version:        12
Bridge OS Version:     3.5 (16P5200)
Anonymous UUID:        52B50B79-B56A-D140-46E9-A1FA6C8EA099

Time Awake Since Boot: 4600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
[... 1484 lines cut ...]
STACK GUARD                        164K       41 
Stack                            136.8M       42 
Stack Guard                       56.0M        1 
VM_ALLOCATE                      133.8M       94 
WebAssembly memory                12.0M        3 
__DATA                            73.6M      427 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       244.7M      134 
__TEXT                           315.9M      401 
__UNICODE                          564K        1 
mapped file                       54.1M      223 
shared memory                      692K       16 
===========                     =======  ======= 
TOTAL                              4.0G     1639 
TOTAL, minus reserved VM space     2.2G     1639 

`

Crash Report by Windows:
-- sorry, could not get crash log

Crash Report by Ubuntu:
`Segmentation Fault`
Not very helpful.

Ghostkeeper, 2019-07-15:
Hmm, according to that stack trace it seems to crash somewhere when connecting QML properties. We could be looking at a genuine Qt bug here.

Just to be sure, we'll need to have Cura's log file as well though. You can find where it's located [here](https://github.com/Ultimaker/Cura#logging-issues).

I'm unable to reproduce the issue on Ubuntu here. What printer model are you switching from/to? What are the reproduce steps?

rmpel, 2019-07-15:
Haven't looked at the contents of the log yet, but here it is :)

[cura.log](https://github.com/Ultimaker/Cura/files/3392483/cura.log)

rmpel, 2019-07-15:
have to note; I reproduced the error at work, not at home, so my printers cannot be reached at this time.
I can say, however, that the software responds identically here as it did at home
-> when on Monitor tab, switching printers crashes Cura
-> when on Prepare or on Preview, switching printers does NOT crash Cura.
It does not matter if a model is loaded on the build area or not.

Ghostkeeper, 2019-07-16:
Seeing some very strange things in that log file, like the `support_roof_enable` setting not existing. It doesn't explain why Cura crashes when you switch printers ONLY in the Monitor stage, but it's still a bit worrying.

rmpel, 2019-07-16:
I've used Cura before (for about a minute, version 15.04), but never on this computer, so I should be able to say; this is the first and only version of Cura on this machine.
I am very new to Cura and have no idea what `support_roof_enable` should do...

thanks for your interest and support!

Ghostkeeper, 2019-07-16:
Version 15.04 doesn't automatically get its settings upgraded to version 2.0+. The version upgrade system was implemented in version 2.1, so your old installation shouldn't affect the new one.

I think the actual crash is somewhere in OctoPrint. We haven't seen anything like this so far.

fieldOfView, 2019-07-17:
Hi, creator/maintainer of the OctoPrint Connection plugin here. This report is also on my radar, but I have not reproduced it yet. 

Are both printers connected to two different OctoPrint instances?

rmpel, 2019-07-18:
Yes, both my printers have a dedicated Pi with OctoPi distro.

fieldOfView, 2019-07-18:
Something to try is to disable showing the camera stream for one or both of the OctoPrint instances (via the Connect to OctoPrint dialog), and then see if switching still crashes Cura. If it doesn't, chances are the crash happens due to the mjpeg streaming code.

rmpel, 2019-07-18:
Will try tonight at home, will report back :) (will be around 18:00 CEST)

fieldOfView, 2019-07-18:
By the way: thanks for the very complete report and the eagerness to work with us on a fix.

rmpel, 2019-07-18:
Pleasure ;) Developer with 28 years on record here. If only I taught myself Python, I could do a lot more ;)

nallath, 2019-07-18:
Good bug reports are worth their weight in gold! So +10 internet points to you.

rmpel, 2019-07-18:
Even without the camera view (double checked on both connections ;) ) Cura still crashes.

rmpel, 2019-07-18:
just to be thorough, the log of the latest crash;
[cura.log](https://github.com/Ultimaker/Cura/files/3407143/cura.log)

fieldOfView, 2019-07-18:
That mostly rules out the camera streaming part of the code then.

One more thing to test: could you disconnect your computer from the network before switching? If it still crashes, that would rule out a race-condition of a network response coming in after the network manager has been destroyed.

rmpel, 2019-07-19:
Would it give the same result when switching the printers when not connected to the home network, like having my laptop at work? In other words; would the problem be "connected to the network the printers are on" or "connected to any network at all" ?
Because the first option has been tried already, resulting in the same problem. 

I will fire up my laptop, disable WiFi and report back (will be a few minutes ;) )

rmpel, 2019-07-19:
Nope, alas, still crashes.

rmpel, 2019-07-19:
I've tested something else
I have 4.0.0 Beta on my computer at work, I added my printers, but of course, no connection to my OctoPis, no problem
I installed 4.1.0, no problem
I added the plugins I use at home (OctoPrint and Sidebar GUI); no problem!
I added (manually) an OctoPrint server; no problem. (might be because I could not authenticate)
Finally I copied my 4.1 folder (from ~/Library/Application Support/Cura/) to my work computer; crashes happen!

So perhaps this will help in finding the problem;

[4.1.zip](https://github.com/Ultimaker/Cura/files/3409915/4.1.zip)

fieldOfView, 2019-07-19:
> I added (manually) an OctoPrint server; no problem. (might be because I could not authenticate)

Two observations: 
* you say you added an instance (one). The crashes happened when switching between two connected instances, right?
* if you don't authenticate, the instances are not connected

After you copied the 4.1 folder to your work computer, were the two instances connected?

rmpel, 2019-07-19:
in below response I make the following assumption:

"connected" means "the printer profile was linked to an octoprint profile"
and not "the network is active, the octopi machines are running and reachable"

> the crashes happened when switching between two connected instances, right?
yes

interesting fact: when removing the OctoPrint instances (Settings > Printers > Connect OctoPrint > remove all instances) the crashing stops.
The plugin for OctoPrint is still active, the settings for the link (API key) is still present.

> if you don't authenticate, the instances are not connected

thinking back; I could not verify an API key, so I guess that would be "not connected"

> After you copied the 4.1 folder to your work computer, were the two instances connected?

hard to say;
the full settings were copied, and the printers were connected (linked to pi) on the source computer so I have to assume they were connected on my work computer.

rmpel, 2019-08-03:
It's been a little quiet here, so I guess you are all very busy, but that's ok, so was I :)

I'd like to report the following:

Cura 4.2.1 + OctoPrint plugin 3.5.8 no longer crashes when switching in monitor tab.

So if you found the bug and solved it; fantastic!!! Thank you!
But if you did nothing specific to this issue; you have new mystery ;)

In any case; thank you very much!

R.

fieldOfView, 2019-08-03:
Thanks for reporting this is likely fixed. I must admit I did not have time to really look in to it, so “mystery” it is. Please keep me posted if the issue recurs (mysterious issues have a tendency to do so)

fieldOfView, 2019-08-03:
For further discussion please see https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/122

closed 2019-08-11 by Ghostkeeper