Ultimaker/Cura #6009: [4.0.0] Rotate->Flat progress goes beyond 100%

10MINT, 2019-07-14:
**Application version**

4.0.0

**Platform**

Windows 10

**Printer**

Custom (Tevo Tarantula)

**Reproduction steps**

Load a geometrically complicated model into Cura and click "Rotate" then "Flat"

**Actual results**

The progress bar goes beyond 100% and does weird transformations, like rotating the object and putting it out of bounds

**Expected results**

The progress should stop at 100% and should only rotate the model

**Additional information**

[image]
Tested with Iron Man mask chin part

Ghostkeeper, 2019-07-15:
Yup, I see it happening. Just need to load a model with a reasonable size otherwise the progress bar is too fast to read. (I was using [this one](https://github.com/Ghostkeeper/SlicerTestModels/blob/master/my/rotary_tumbler.scad).)

BagelOrb, 2019-08-06:
Still happens in 4.2.1
[image]

BagelOrb, 2019-08-06:
At 100% the object is already laid flat, but the loading bar continues for no apparent reason.

closed 2024-11-25 by GregValiant