Ultimaker/Cura #6010: Crash when expanding Support or Build Plate Adhesion setting sections

cyber-murmel, 2019-07-14:
**Application version**
master built today (2019-07-14)

**Platform**
- Lenovo X270
- Linux x270 5.1.16-1-MANJARO #1 SMP PREEMPT Thu Jul 4 20:32:22 UTC 2019 x86_64 GNU/Linux
- Intel Corporation HD Graphics 620 (rev 02)
- video-linux driver

**Printer**
Anycubic Kossel Linear Plus

**Reproduction steps**
1. Open Cura
2. Expand "Print settings" on the top right
3. Switch to "Custom"
4. Expand "Support" section

**Actual results**
The section doesn't expand, cura stalls for a second and crashes.

**Expected results**
Expansion of the section and no crash

**Additional information**
The log doesn't contain all the output of the program. This is the missing part:
```
2019-07-14 16:15:59,750 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2019-07-14 16:16:21,719 - WARNING - [MainThread] UM.Decorators.deprecated_function [19]: <function MachineManager.activeMachineId at 0x7f3de8561d90> is deprecated (since 4.1): use Cura.MachineManager.activeMachine.id instead
/usr/lib/python3.7/site-packages/UM/Qt/QtApplication.py:341: DeprecationWarning: <function MachineManager.activeMachineId at 0x7f3de8561d90> is deprecated (since 4.1): use Cura.MachineManager.activeMachine.id instead
  super().exec_(*args, **kwargs)
Fatal Python error: Segmentation fault

Thread 0x00007f3daddbc700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f3dae5bd700 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f3dbbfff700 (most recent call first):
  File "/usr/lib/python3.7/site-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f3ddcaeb700 (most recent call first):
  File "/usr/lib/python3.7/site-packages/UM/Backend/Backend.py", line 138 in _storeOutputToLogThread
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f3dde7fc700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f3ddeffd700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f3ddf7fe700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f3ddffff700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Current thread 0x00007f3df2e90680 (most recent call first):
  File "/usr/lib/python3.7/site-packages/UM/Qt/QtApplication.py", line 341 in exec_
  File "/usr/lib/python3.7/site-packages/cura/CuraApplication.py", line 799 in run
  File "/usr/bin/cura", line 137 in <module>
[1]    24698 segmentation fault (core dumped)  cura
```

cyber-murmel, 2019-07-14:
I've also added a new printer, to see whether it correlates with my printer settings. Problem remains the same.

Sphiinx, 2019-07-15:
Duplicate issue: #6008

nallath, 2019-07-15:
Closing this, since it's a duplicate.

closed 2019-07-15 by nallath