Ultimaker/Cura #6014: Feature Request: Concentric rib and/or gusset infill

sobhambi, 2019-07-15:
**Is your feature request related to a problem? Please describe.**
Current infill profiles make it challenging to print fast while getting the needed strength for some models. Profiles that guarantee strength tend to use quite a bit of material and their movements increase print time significantly. Profiles which are easier on print times with some strength are are quite inconsistent with their placement on some models.

**Describe the solution you'd like**
Adding ribs/gussets to a part would make it stronger overall or in specific areas. Since its concentric it conforms to the shape of the part (its internal structure) and one can determine if it should be on corners, holes, vertical/horizontal or both, width, height, density, shape, inside or outside the part, connected to outer/inner wall etc.

**Describe alternatives you've considered**
Manually inserting ribs/gussets into some solid models maybe difficult without ruining the model. Especially if its a borrowed model rather than a self-created one (if that makes sense).

**Affected users and/or printers**
Those looking for an economical/material friendly way of achieving consistently stronger walls without spending hours trying to manually place reinforcements into the model.

**Additional context**

[image]
[image]
[image]
[image]
[image]

nubnubbud, 2019-07-16:
While those kinds of ribs are definitely good for part strength, the fact that there are gaps between layer lines makes them a bit less effective at their function than in injection molding. Not only are layer lines weak along the main direction force will be applied (tensile and buckling compression at edges) infill and similarly thin structures are notoriously brittle in 3D printable materials, and are mostly for ensuring top surfaces have something to bridge over well.

also of note is that these kinds of parts often go through structural analysis in order to determine the best locations for the ribbing. How does the computer know where it's needed? I don't see this kind of thing being automated well. It's one thing that humans are just better at doing manually. you can't expect a computer to make smart choices, because a computer typically has no clue what "strong" or "useful" is.

the hardest part is the algorithm though. I would personally select a maximum height from the build plate, and that's where the triangles start. each one is placed along the face's negative normal at regular intervals, or to a minimum of 4 at even intervals, to include pins and rods. it's then sliced with those triangles as part of the model.

you could also only leave infill a certain distance from the sides of parts. if you combine that with multiple extrusion-thick infill, you can get something similar to this effect even with current settings!

_**However! I (in a completely personal sense) think this kind of feature belongs in the cad program/designer you use to make the model. A slicer should be rather portable. It's not to change the shape of an object, but rather make it a printable file. Most features should enable new models/materials to be printed in a better way. What you're suggesting reminds me of a modifier in blender, where you apply an algorithm to prodecurally change the model. maybe look there, first. As a general rule, I never bring someone else's STL file into cad, but blender. When it's turned into an STL mesh, mesh editing becomes the best way to work with it. Keop going! once you're good at mesh editing, gussets and ridges should be doable in no time at all!**_

Ghostkeeper, 2019-07-16:
I could hardly have said it better than Nubnubbud. The main problem is that this is very hard to automate, and Cura is not the sort of program where people expect to manually modify this. We leave this to CAD.

The algorithmic part is hard because it pretty much requires 3D algorithmics. This makes it orders of magnitude harder to implement; there are very few known algorithms that work properly on surface meshes, especially if they are not 100% manifold. This is part of the reason why we've purposefully set all of those CAD-like-alterations outside of Cura's scope.

So I'm closing this because it's out of scope and not likely to be brought in scope due to its difficulty.

closed 2019-07-16 by Ghostkeeper