Ultimaker/Cura #6015: Cura 4 & 4.1 Creates random vertical planes in prints

Cj1190, 2019-07-16:
Cura 4 & 4.1 Create random vertical planes that intersect printed geometry or floating in random space. This does not occur with Curra 3.6 when the exact same stl files are used. This has caused every print with Cura 4 an 4.1 to fail. Have reinstalled and tried to force them out. Sometimes they show in cura, sometimes they do not. Problems occured in a non-moded version of Cura and on two different computers. Your format is below. Also the new... interface is horrid...

**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

Ghostkeeper, 2019-07-16:
I don't see that printed vertical geometry as you describe. Can you share a project file with us, or some reproduce steps? Or a screenshot of what you think is horrid about the interface? Our format for bug reports is not just for show...

Ghostkeeper, 2019-07-16:
Made with Cura 4.1. I don't see any vertical planes, random extrusions in space or incorrect extrusions throughout geometry.

[image]

closed 2020-06-19 by no-response[bot]