Ultimaker/Cura #6019: Progress percentage on the software do not match the display on the CR-10S

CGLee65, 2019-07-16:
The Completed % on the software shows like 4% and the Display on the control box says 0%

rburema, 2019-07-17:
There's some known (and unkown) issues with progress reporting / time estimation / etc. One thing to keep in mind is that it's an estimation, especially for printers we can't often test with.

Does this discrepancy become wildly out of sync or does it stay within certain limits?

CGLee65, 2019-07-17:
I figured out that the percentage does not show on the control box when printing via USB cable, but it does show on the monitor screen of cura.  The percentage does show on the box when you print from the TF Card and does not show anything on the monitor page.

From: Remco Burema
Sent: Tuesday, July 16, 2019 23:31
To: Ultimaker/Cura
Cc: CGLee65; Author
Subject: Re: [Ultimaker/Cura] Progress percentage on the software do not matchthe display on the CR-10S (#6019)

There's some known (and unkown) issues with progress reporting / time estimation / etc. One thing to keep in mind is that it's an estimation, especially for printers we can't often test with.
Does this discrepancy become wildly out of sync or does it stay within certain limits?

Ghostkeeper, 2019-07-17:
The printer doesn't know the progress when printing via USB cable. The software is sending the g-code commands on the go.

There is no way for Cura to communicate progress to a CR-10S via USB cable. The only g-code that Cura could use for that is M532 but that's only supported for Repetier, according to the [RepRap wiki](https://reprap.org/wiki/G-code#M532:_Set_print_progress).

yet-another-average-joe, 2019-07-17:
Marlin has M73 that displays a percentage on the LCD :

http://marlinfw.org/docs/gcode/M073.html

(it is used by a couple of OctoPrint plugins)

fieldOfView, 2019-07-17:
M73 support needs to be specifically enabled in Marlin, so it will not work on all printers.

This feels like something that maybe should be implemented in a postprocessing plugin, calculating progress from the ";TIME_ELAPSED" comments that CuraEngine puts into the gcode after each layer.

yet-another-average-joe, 2019-07-17:
I was just thinking of the Cura Post Processing Plugin, and realized that it could be based on layer # only, and it does not make sense !

I thought that LCD_SET_PROGRESS_MANUALLY was ON by default. I was wrong...

LinFor, 2019-10-03:
I wrote simple post-processing script to do thins on Marlin: https://github.com/LinFor/Marlin/blob/e-bot-skr/cura-profiles/scripts/EnrichPrintProgress.py
It's just use TIME_ELAPSED layer estimation from Cura and converts it into M73/M117 commands. Maybe later I approximate progress more precisely within layer (by E-movements, for example) to do more reliable progress estimation for short prints or very big layers.

Also, currently Marlin ignores M73 command when printing from SD card (so it's the only reason to produce M117 commands), I wrote PR for Marlin to fix this, please vote for my PR if this functionality is important to you: https://github.com/MarlinFirmware/Marlin/pull/15449

closed 2023-09-29 by github-actions[bot]