Ultimaker/Cura #6020: Visualize Printing in Realtime

CGLee65, 2019-07-16:
Can you add the ability on the Monitor screen to visualize the printing layers as they are printed like the preview screen but in real time.  This would really help to locate errors so they can be fixed in the print.

rburema, 2019-07-17:
This heavily depends on the information the printer sends back, which, in most cases, is not a lot. The best we can do in general would be to slowly run similation view in sync with the time estimate, which might be off, like you observed in the last ticket.

CGLee65, 2019-07-17:
Ok, I am Running a CR-10S 300.  I know with my old Tevo I could watch the print on my PC that was connected via USB.  I thought Cura Did the same thing.

From: Remco Burema
Sent: Tuesday, July 16, 2019 23:42
To: Ultimaker/Cura
Cc: CGLee65; Author
Subject: Re: [Ultimaker/Cura] Visualize Printing in Realtime (#6020)

This heavily depends on the information the printer sends back, which, in most cases, is not a lot. The best we can do in general would be to slowly run similation view in sync with the time estimate, which might be off, like you observed in the last ticket.

rburema, 2019-07-17:
Unless the information becomes standardized, we'd have to redevelop that for each type of printer.

Ghostkeeper, 2019-07-17:
When printing via USB, Cura is sending out the commands to a serial port almost in real time. There's just a few commands of delay in the printer. It would theoretically be feasible.

When printing via Wifi or OctoPrint, it'd be very hard to keep it in sync because of the 5% variation in clock speed on most of those boards. You'd need some sort of protocol for very precise (line-number-in-gcode-precise) progress reporting.

All of these have the problem that we don't normally show all of the g-code in the preview stage. We show the part of it that CuraEngine generates. The rest of it, like start g-code and end g-code, extruder switching g-code and such, are not shown unless you save and load the g-code back in which takes a few minutes. And part of the consequences of that is that we currently can't make a link from certain lines of g-code to where the printer is in the progress of the layer view.

And then there is of course the political side that Ultimaker doesn't want to spend time on improving USB cable printing support. If it's a small amount of time we can probably squeeze it in somewhere, but these hurdles are big enough that we don't want to start it. That's why I'll defer this now for when there are no other more pressing things to work on. If we find a way to implement this via the network, that makes it more valuable for the political side.

ardent42c, 2020-01-18:
This feature is implemented very nicely in Repetier Host (been using it years ago), so It's obviously possible.

VladekRo, 2021-03-23:
The same here. Had to go back to Repetier Host (Repetier Server is unreliable in my Win10 Insiders) and use Cura as an external slicer.

jaggzh, 2021-05-20:
Might I ask why's it such a big deal? Seems like literally one or two lines of code.  The printing can just set the Preview layer height to the same as the present height of the outgoing gcode no?  If there are other types of prints (like sequential objects), then just add that in there too (or say "does not yet work for 'one-at-a-time' print modes").

VladekRo, 2021-05-20:
Hi
Lack of that small feature is like a pinch of salt on otherwise great cake 😢

Growing Old is Compulsory. Growing Up is Optional.

Democracy: Where two idiots outvote a genius???

Ghostkeeper, 2021-06-02:
> Might I ask why's it such a big deal? Seems like literally one or two lines of code. The printing can just set the Preview layer height to the same as the present height of the outgoing gcode no? If [...]

If you'd like to write those two lines, you're welcome to!

I think what you're thinking of is that every time Cura sends a command over the USB cable, it would increment the layer view progress by 1 line. Aside from details like how it should be set up correctly when the print starts, etc. that would indeed be not so hard.

If you'll implement that though, you'll quickly find that:
* A single g-code line is not equal to a single line in the layer view.
* Not all lines are visualised in layer view. In fact, not even all extruded, printed lines are visualised in layer view.
* When the line is sent to the printer is not actually when the printer executes the g-code line. The printer is typically 10-20 seconds ahead. Sometimes up to 2 minutes, for big first layers.
* Updating layer view 200 times per second is not a good idea, and will ruin your print quality by taking the Python GIL.

All of these are less of an issue for just updating the layer number though. A layer in the layer comments is almost always equal to a layer in the layer view (the only exception being spiralize). They are all visualised except extra layers added via post-processing scripts. The slower time scale makes a minute delay less of an issue. And you'd update at most every 10 seconds or so which is no problem at all.

Yet all of these solutions have one other problem: It's not gonna work for wifi or cloud printing. And those two are the only ones that Ultimaker currently uses in their printers. As a result, I can't spend time on it to implement features for USB cable printing for you, unless I'd implement it in my free time. I do those things sometimes, but I have other things I find fun as well.

VladekRo, 2021-06-02:
Hi
Thank you for reply.
Best Regards
Vlad
Darwin, NT, Australia
PS
My time is running out too, so I don't think I can find time to fiddle with your code.

Growing Old is Compulsory. Growing Up is Optional.

Democracy: Where two idiots outvote a genius???

jaggzh, 2021-06-10:
How does the wifi printing (and cloud) send gcode?  A whole gcode file at a
time?

Ghostkeeper, 2021-06-10:
> How does the wifi printing (and cloud) send gcode?  A whole gcode file at a
time?

Yes, Ultimaker's implementations of wifi and cloud printing do. And all other implementations that I know of too, in fact. This is more reliable, since a wifi connection could be disrupted during the print.

elitewolverine, 2021-11-18:
> > Might I ask why's it such a big deal? Seems like literally one or two lines of code. The printing can just set the Preview layer height to the same as the present height of the outgoing gcode no? If there are other types of prints (like sequential objects), then just add that in there too (or say "does not yet work for 'one-at-a-time' print modes").
> 
> If you'd like to write those two lines, you're welcome to!
> 
> I think what you're thinking of is that every time Cura sends a command over the USB cable, it would increment the layer view progress by 1 line. Or every time you encounter `;LAYER` to increment the layer number. Aside from details like how it should be set up correctly when the print starts, etc. that would indeed be not so hard.
> 
> If you'll implement that though, you'll quickly find that:
> 
> * A single g-code line is not equal to a single line in the layer view.
> * Not all lines are visualised in layer view. In fact, not even all extruded, printed lines are visualised in layer view.
> * When the line is sent to the printer is not actually when the printer executes the g-code line. The printer is typically 10-20 seconds ahead. Sometimes up to 2 minutes, for big first layers.
> * Updating layer view 200 times per second is not a good idea, and will ruin your print quality by taking the Python GIL.
> 
> All of these are less of an issue for just updating the layer number though. A layer in the layer comments is almost always equal to a layer in the layer view (the only exception being spiralize). They are all visualised except extra layers added via post-processing scripts. The slower time scale makes a minute delay less of an issue. And you'd update at most every 10 seconds or so which is no problem at all.
> 
> If you'd implement that with the `;LAYER` comment it would only work on Cura-generated code. If you'd implement that with Z changes you need to be very careful with things like Z hop and spiralize.
> 
> Yet all of these solutions have one other problem: It's not gonna work for wifi or cloud printing. And those two are the only ones that Ultimaker currently uses in their printers. As a result, I can't spend time on it to implement features for USB cable printing for you, unless I'd implement it in my free time. I do those things sometimes, but I have other things I find fun as well.

Couldn't it be...as it sends the code to the printer...the video display simply graphs the gcode.  So while you are not getting a live feedback you are getting a 'preview' of what should be printing within the next...0.1seconds or half a second.

Even that is better than looking at a white screen.

Use the build plate as the 'plane', use a generic 50polygon nozzle model, translate gcode moves into xyz coordinates, using the buildplate numbers.  And boom?

I used to code back in the day, nowhere near enough to do itmyself but i mean this is 1990s type of code?

Ghostkeeper, 2021-11-25:
> Couldn't it be...as it sends the code to the printer...the video display simply graphs the gcode. So while you are not getting a live feedback you are getting a 'preview' of what should be printing within the next...0.1seconds or half a second.
> 
> Even that is better than looking at a white screen.

The printer is typically 10-20 seconds ahead, sometimes up to 2 minutes, for big first layers. Now that may be something that users would accept. Or maybe not. It's hard to say.

The first two problems are, I think, the most difficult to solve. There is no 1-to-1 mapping between the g-code and the layer view data that CuraEngine generates for visualising. A mapping can certainly be created, but it's not as easy a solution as you'd make it out to be.

And then there's my final point too:

> Yet all of these solutions have one other problem: It's not gonna work for wifi or cloud printing. And those two are the only ones that Ultimaker currently uses in their printers. As a result, I [...]