Ultimaker/Cura #6021: Need to retract material when traveling on base layer

lukehutch, 2019-07-16:
**Application version**

4.1.0

**Platform**

Linux

**Printer**

Anycubic i3 Mega

**Description**

When printing on an Anycubic i3 Mega, long-distance travel on the base layer often continues to extrude material at about half the normal rate (or maybe it is that the head is moving at twice the normal speed). This leaves lines in the printed material. It ruins the regularity of the visual finish of the base layer. I see this on other layers too, but the visual finish of the base layer is most important, since for a lot of parts, this is a user-facing face after the part is printed.

This mainly seems to happen across solid regions in the base layer, not between disconnected components. Travel between disconnected components appears to retract the material correctly.

The extra extruded material is not visible in the Cura preview mode, so maybe it is a bug in the gcode generation?

The issue is present in all prints, so I have to believe this is a known behavior.

smartavionics, 2019-07-16:
Hello @lukehutch, try setting combing mode to no skin and the max comb distance without retract to something like 20. If that doesn't fix your problem, please save the project (File -> Save), rename or zip the .3mf file and attach to this issue. Thanks.

lukehutch, 2019-07-16:
I just discovered "Combing Mode -> Within Infill", and came back here to close the bug. That's what the issue was, as far as I can tell. Thank you!

lukehutch, 2019-07-16:
PS maybe "Combing Mode" should be set to "Within Infill" by default. This is a much more sensible default, I think.

closed 2019-07-16 by lukehutch

smartavionics, 2019-07-16:
No, within infill works very badly, you need to use no skln mode.

lukehutch, 2019-07-16:
Thanks. Maybe "Combing Mode" should be set to "Not in skin" by default? This is a much more sensible default, since it increases the visual quality of the prints.

Ghostkeeper, 2019-07-17:
Visual quality is one thing to optimise for.

It's not set to "Not in Skin" either because it causes longer travel moves as well as more retractions. More retractions cause more grinding of the filament, reducing reliability in long prints due to the feeder losing grip. Both effects lead to longer printing time.