Ultimaker/Cura #6022: Cura 4.0 head wandering error - no problem with same file in 2.7

rickcad, 2019-07-16:
[CCR10S_Shelfie-pinned-Assembly (1).3mf.zip](https://github.com/Ultimaker/Cura/files/3399304/CCR10S_Shelfie-pinned-Assembly.1.3mf.zip)
I made this part on my CTC printer using replicatorG
I bought a new Artillery X1 and printed several other items successfully using Cura 2.7 and 4.0.
I then tried to print this part using Cura 4.0. It printed several layers and then the head went all the way to the right and was moving around strangely.  I thought it was the new printer acting badly. I tried a number of orientations and even exported the file again to an .stl and redid everything with the same bad results. Then I loaded up Cura 2.7 and ran it there. It printed perfectly, so there is a bug in Cura 4.
Regards,
Richard

**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

rburema, 2019-07-17:
I see you're using a Creality profile for your non-Creality printer? It might be that something changed in that profile that makes it incompatible with your printer.

(In that case:) Perhaps you could try the 4.2-BETA, since we pulled in the mod previously known as Creawsome, which changes a bunch of those profiles again.

rickcad, 2019-07-17:
Good point. When I went to the Cura 2.7.0 I used other for the printer and added the setup info for the X1. I put the correct parameters in the creatlity printer setup, but I will create an "other" printer in Cura 4.0 and see if the same thing happens. Thank you, Rburema, for your reply!

closed 2020-06-19 by no-response[bot]