Ultimaker/Cura #6023: 4.1 is corrupting SD cards

xanuser, 2019-07-16:
I have Fatshark FPV goggles that record video on micro SD card. If that card is left connected to my windows 10 PC when cura 4.1 has a sliced file ready to "save to SD card" the card is corrupted and must be fixed with chkdsk in CMD prompt and reformatted..
Version of cura prior to 4.0 did not have this same issue at all.
This happens even if I don't save any gcode to the SD.
I Have rolled back to 3.6.0 to eliminate the problem.

did y'all change something about how cura handles SD cards in 4, or is this a bug, bug?

**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

rburema, 2019-07-17:
The only thing we can think of is the USB-printer-detection.

You could try turning this off:
 - go to the marketplace (button top right)
 - go to installed plugins
 - uncheck USB printing
 - press the quit cura button

justin-sprecher, 2019-10-12:
This bug is still an issue. I also cannot print via usb anymore. Rolling back to 3.6 now.

Ghostkeeper, 2019-10-14:
We have been unable to reproduce this problem. I've been printing via SD cards fine and USB cable as well. Something must be different about our set-ups. The only way that we can try fixing the issue is by identifying what is different.

closed 2020-06-19 by no-response[bot]