Ultimaker/Cura #6025: Adding controls to allow USB access through Cura

Andy3820, 2019-07-17:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Affected users and/or printers**
 
**Additional context**

Andy3820, 2019-07-17:
My older printer (Cocoon Create-Australia/Wanhao rebadged) doesn't have wifi and I'd love the new versions of Cura to allow the use of a USB connection. This would probably benefit a large number of people with older printers and cut down the time-consuming steps of creating gcode files, copy to SD card, insert into the printer, select from the touch screen on the printer and so forth.

closed 2019-07-17 by Andy3820

Ghostkeeper, 2019-07-17:
This was implemented in version 12.07, seven years ago, the very first release of Cura! :tada: