Ultimaker/Cura #6026: [4.1.0] Unable to print again until printer has cooled down

dryphi, 2019-07-17:
Hi, I'm having trouble printing via USB when I have just finished another part. It seems I have to wait until the printer is completely cooled down before it will let me print again. The previous part is listed at 100% complete on the Monitor tab. I have removed the previous part. I click the Prepare tab and click "Print via USB" and nothing happens. I don't get an error popup or anything, it just does nothing at all. Finally after the printer has cooled down and the "Finish!" message appears on the printer screen, then I can click "Print via USB" on Cura and it will print again. It would be nice if I could print immediately after another print has finished so I don't have to wait for the printer to cool down and heat back up again.

This is for Cura 4.1.0 on an MP Select Min v1 on Windows 10 x64 printing via USB.

The part file is irrelevant it happens with every part. This happened with previous versions of Cura as well I just wasn't bothered enough to report it.

Might be similar to bug #1110 but I think it's a separate issue.

[cura.log](https://github.com/Ultimaker/Cura/files/3403925/cura.log)

**Application version**
Ultimaker Cura 4.1.0

**Platform**
Windows 10 x64

**Printer**
MP Select Mini v1

**Reproduction steps**
Every time I print and want to print again immediately after it's finished and I've removed the previous part.

**Actual results**
When I click the "Print via USB" button nothing happens. No error, nothing, it just sits there. I have to wait until the printer is completely cooled down then when I press "Print via USB" it goes to the MONITOR tab and prints like it's supposed to.

**Expected results**
I should be able to print when the printer is still warm so I don't have to wait.

**Additional information**

smartavionics, 2019-07-18:
The problem is probably due the end g-code defined in the Malyan M200 printer definition that the MP Select definition is based on. At the moment, the end g-code looks like this:
```
G0 X0 Y127;(Stick out the part)
M190 S0;(Turn off heat bed, don't wait.)
G92 E10;(Set extruder to 10)
G1 E7 F200;(retract 3mm)
M104 S0;(Turn off nozzle, don't wait)
G4 S300;(Delay 5 minutes)        <------- 5 minute delay!
M107;(Turn off part fan)
M84;(Turn off stepper motors.)
````

There's a 5 minute delay in there! To stop it doing that, edit the end g-code in the printer settings dialog and comment out or remove the last three lines. You will then have to turn off the part fan and/or the steppers manually when you have finished printing. Hope this helps.

Ghostkeeper, 2019-07-18:
So the confusion here arises from Cura indicating that it's 100% complete, which is a rounded number based on the lines of g-code it has processed. Since it's practically at the end, it rounds to 100% but there is a command at the end that takes way longer than the other commands.

Also, we could consider removing the Print Via USB option temporarily while another job is still printing. Or putting an error up if you try to print two things at the same time.

Ghostkeeper, 2019-07-18:
I've made it show a message if you try to start another print. Hopefully that makes things a bit more clear.

closed 2019-07-18 by Ghostkeeper

dryphi, 2019-07-18:
Okay great thanks for all the support.
How do I get to edit G-code in the machine settings? I can't find that option. I've edited G-code in previous versions (e.g. 15.04.6) but not in this new layout. It doesn't seem to be anywhere in the settings menu.

smartavionics, 2019-07-18:
Go to Settings -> Printer -> Manage Printers, it's in the printer settings dialog like this example...

[image]

Ghostkeeper, 2019-07-19:
You can still send g-code directly. Only when printing via USB though. You'll find it in the monitor stage.

commit cc960973 (2019-07-18): Show a message when trying to print via USB while already printing

Otherwise the button does nothing and gives no feedback of why it does nothing.

Fixes #6026.