Ultimaker/Cura #6027: Please add your app as a flatpak

Jacalz, 2019-07-18:
**Is your feature request related to a problem? Please describe.**
Linux is a hard beast when it comes to getting the application to work on multiple different distros and right now you seem to solve that by appimages. While they do make the apps usable on multiple distros they do come with flaws compared to alternative formats. For example the lack of simple/automatic updates and no sandboxing.

**Describe the solution you'd like**
I would love to see this app made avaliable as a Flatpak on https://flathub.org as it would be a simple way of distributing the application in a fully open source format (compared to snap which is not oss) and it provides users with simple updates from the command line or in the software shop. It also has the great power of sandboxing the application for an extra layer of security to the users.

**Describe alternatives you've considered**
One could do it using the Snap format, but it is not fully open source, has bad theme integration, is terribly slow to start apps and has low adoption outside of Ubuntu. I find flatpak to be better in every way.

**Affected users and/or printers**
- Almost all Linux users using the AppImage (except for the small cases where flatpak isn't supported or when a user wants to use an old version deliberately).

**Additional context**
https://github.com/flathub/flathub/blob/master/CONTRIBUTING.md

Ghostkeeper, 2019-07-19:
Sorry, we'd like to continue providing the AppImage ourselves. We count downloads in order to have a statistic of how many users are using Cura vs. how many crash reports we get and such, to help determine if the crashes warrant a patch release or not. FlatHub doesn't allow us to do that.

closed 2019-07-19 by Ghostkeeper

probonopd, 2019-07-31:
@Jacalz  AppImage is specifically designed so that
* The author of an application is in full control over the whole end-to-end software distribution experience with no gatekeepers nor middle-men services in between.
* The user can keep many different versions alongside each other
* The application can be managed without interfering with the operating system's package manager, and can never change the system

@Jacalz what advantage would you think would Flatpak give you? Could the same advantage be achieved by improving AppImage?

Jacalz, 2019-07-31:
There is no point in stopping the creation of the appimages just because it’s packaged as a Flatpak, but I see the you don’t want Flathub to be in control over your app so to say. I think there is a better way to do it if you want full control and I think that could be managed by having your own Flatpak repo. An example of this would be Citra, see their Flatpak page [here](https://flatpak.citra-emu.org/). 

What Flatpak does miles better than Appimages are the nature of how they are self updated (if you have automatic updates on, otherwise just run `flatpak update` from cli ) and they allow easier updates for users without having to download a new appimages every time. It also allows installation without interfering with system packages or system files, just like Snaps and Appimages. Another great thing is that the dependencies are part of a Flatpak repo and can be updated besides packages. This makes them more secure than appimages as you can keep them
updated.

I am NOT talking about removing AppImages, just about adding Flatpak support :)

probonopd, 2019-07-31:
> What Flatpak does miles better than Appimages are the nature of how they are self updated

Are you aware of AppImageUpdate binary delta updates? Ultimaker could choose to embed update information into the Cura AppImage, or even embed the updater, making the AppImage "self-updateable".

> Another great thing is that the dependencies are part of a Flatpak repo and can be updated besides packages. 

With AppImage, all depenencies that cannot be expected to come with the base system (default installations of the major Linux distributions) come bundled within the AppImage, so that the user never has to download something in addition to the AppImage.

> I am NOT talking about removing AppImages, just about adding Flatpak support :)

Yes, I know, I just wanted to understand better why anyone might want that when there is an AppImage.

Ghostkeeper, 2019-08-02:
We're fine with people making their own distributions though. It's unnecessary for us to maintain both. AppImage has support for a far wider range of use cases than we'll ever need so we're quite happy with it so far.

JamieCarlAPIR, 2021-04-28:
I came here while looking for the state of Cura and Flatpak and just wanted to note that Cura is in the main Ubuntu APT repository (https://packages.ubuntu.com/groovy/cura) and available as a Snap from Snapcraft (https://snapcraft.io/install/cura-slicer/ubuntu).  The Ubuntu APT is a little out-dated at version 4.6.2 while 4.9 is the current release at the time I'm writing this.  The snap appears to be current.

So the reason given here by @Ghostkeeper about keeping with AppImage appears to be invalid, given that MOST Linux users are probably going to use the APT or Snap versions because it's in the Ubuntu Software Centre and Snap Stores (that's what I did).

I understand the desire to collect download statistics, but Cura has the ability for people to "log in" so why not use that data to gather far more accurate usage statistics?  

Surely this argument is now out-dated and we can revisit the possibility of getting a Flatpak?

JamieCarlAPIR, 2021-04-28:
> > I am NOT talking about removing AppImages, just about adding Flatpak support :)
> 
> Yes, I know, I just wanted to understand better why anyone might want that when there is an AppImage. So thank you for answering.

Launcher icon?

probonopd, 2021-04-28:
@JamieCarlAPIR: With AppImage you have a single file you can copy around multiple computers (great for e.g., schools/labs that are not connected to the internet), and most of all: you can keep all versions of Cura in parallel, making it risk-free to try out the latest version.

> Launcher icon?

What about it?
Please open a ticket with your your desktop (Gnome, KDE Plasma, Xfce,...) that you would like to see native AppImage support there. In the meantime, there are plenty third-party solutions to integrate AppImage icons and menu entries:

https://github.com/AppImage/awesome-appimage#desktop-integration

JamieCarlAPIR, 2021-04-28:
> @JamieCarlAPIR: With AppImage you have a single file you can copy around multiple computers (great for e.g., schools/labs that are not connected to the internet), and most of all: you can keep all versions of Cura in parallel, making it risk-free to try out the latest version.

Thanks, but I use a laptop to solve my portability issues.  I don't need you guys to force your portability mentality on me. 

> > Launcher icon?
> 
> What about it?

Seriously?  Ok sure, you're weird.  

Anyway, I honestly don't care what you guys do.  Stubborn developer's gonna be stubborn.  I just thought it was funny that you're all hell bent on AppImage when I can tell you, "normal people" don't use/understand/care/know about them.  Your mentality is why Linux isn't on more desktops.

Ghostkeeper, 2021-05-13:
By all means, you're free to redistribute it on Snappy, Flatpak, Apt, Dpkg, RPM, Yum, Zypper, whichever you fancy, as long as it adheres to Cura's license.

All distribution methods have their advantages and disadvantages. We chose one that is common and has the features we like. And we'll maintain only one, because we don't have the time to maintain all of them. If we switch we'll inevitably get a request to switch back to AppImage or to switch to another because everyone feels strongly about these things and brings up arguments why one is better than another.

egore, 2021-12-08:
In case someone is still interested in how to use Cura from a flatpak installation, see https://github.com/Tim453/cura-flatpak

This unpacks and repackages the AppImage downloaded from github so building this image will still allow counting downloads while hosting it in the flatpak environment.

Ghostkeeper, 2021-12-22:
We're not really paying a lot of attention to counting downloads from Github, as far as I know. But unpacking and re-packaging the AppImage may be a reliable way for you to stay up to date, and ensure traceability.