Ultimaker/Cura #6031: Incorrect UI description for Ironing Feature

dstulken, 2019-07-19:
**Application version**
Cura 4.1.0

**Platform**

(All?)

**Printer**

(Doesn't matter)

**Reproduction steps**

Hover mouse cursor over the "Enable Ironing" setting to view the description

**Actual results**

Cura 4.1.0's misleading description of Ironing (bolding emphasis is mine):
> "Go over the top surface one additional time, **_but without extruding material_**.  This is meant to melt the plastic on top further, creating a smoother surface."

**Expected results**

We would expect the ironing description to reflect the actual operation of the feature, something to the effect of:
> "Go over the top surface one additional time, **_while extruding a small amount of additional material_**.  This is meant to melt the plastic on top further, **_while also filling any remaining small gaps_**, creating a smoother surface."

(The "Ironing Flow" parameter defaults to 5%, so extrusion *is* enabled by default with Ironing.)

**Additional information**

See attached screenshots of misleading description in Cura software, and the more accurate feature description from the Cura 2.7 release announcement.

[image]

[image]

Vandrasc, 2019-07-22:
Hi @dstulken , thank you for your feedback, we will have a look and try to improve the description.

Ghostkeeper, 2019-07-22:
I agree. Fixed it here: https://github.com/Ultimaker/Cura/commit/eca8f3ab577a3157db72137cf8bc7328eca85dd8 .

closed 2019-07-22 by Ghostkeeper

dstulken, 2019-07-22:
Looks good, thanks!

commit eca8f3ab (2019-07-22): Correction to ironing tooltip

Fixes #6031.