Ultimaker/Cura #6032: Printer Pauses with Waiting for user...

CGLee65, 2019-07-20:
Hi, all of a sudden when printing from Cura via USB to my CR-10S the printer pauses and the screen on the printer says "Waiting for User..."  Then if you click the control knob it starts printing again.  This happens multiple times and I can't figure out why it is doing it.

fieldOfView, 2019-07-20:
Please fill in the issue template. We need more information, like what version of Cura you are using.

CGLee65, 2019-07-20:
I am using Cura ver 4.1.0 on a windows PC running Windows 10.

CGLee65, 2019-07-20:
I figured out the issue, my USB ports where shutting off for power saver.  I turned that off and no more pausing.

CGLee65, 2019-07-21:
Hi, Well I thought I solved the issue but it is still pausing and giving the message "Waiting for User..."

Ghostkeeper, 2019-07-21:
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/5989.

It appears to be caused by a recent Windows update then.

closed 2019-07-21 by Ghostkeeper

shikshik, 2019-07-24:
Hey, I had similar stops with Ender 3 Pro + mac, I figured at first the issue was similar, so I checked "Prevent computer from sleeping automatically when the display is off" on the mac power settings, but now I just got the same issue, the only thing left is to uncheck "Put hard disks to sleep when possible".
I do think it could be solved with some settings of Cura installation on the mac (like a video player running in full screen to prevent sleeping while watching a movie).

If it is an option,  it should be opened as a feature / fix.

Cheers!