Ultimaker/Cura #6033: 4.2 Beta fails to import 4.1 curaprofiles

mjf55, 2019-07-20:
**Application version**
4.2 Beta

**Platform**
I'm on Windows 10 Lenovo Thinkpad W520 but does not matter

**Printer**
Creality Ender 3

**Reproduction steps**
0- Make sure you have some ( yours or others ) 4.1 .curaprofile that you want
1- Install 4.2 Beta.  No other Cura Versions on the machine
2- Install a Creality Ender 3 printer via the wizard ( probably any creality will do)
3- Once everything is ready, attempt to import a backup of your 4.1 curaprofile.
4- It will fail

**Actual results**
Error pop up saying 
The machine defined in profile C:/blah/Ender3/Cura4.1.curaprofile (fdmprinter) doesn't match with your current machine (creality_base), could not import it.

**Expected results**
Import successful

**Additional information**
I examined both the 4.1 and created a 4.2 .curaprofile.  What I see different is that Cura changed quality type names from draft, fast, normal, etc (4.1) to low, standard, dynamic, super (4.2) and the definition went from fdmprinter (4.1) to creality_base ( 4.2)

To me it looks like the creawsome changes were pulled into 4.2 beta, but making it incompatible with older non-creawsome profiles. 

[stderr.log](https://github.com/Ultimaker/Cura/files/3413367/stderr.log)
[curaprofiles.zip](https://github.com/Ultimaker/Cura/files/3413368/curaprofiles.zip)

Ghostkeeper, 2019-07-22:
Yeah, we've discovered this as well in our tests. We've made a fix for it that will be released with the stable 4.2 release. Thanks for testing!

Fix is here: https://github.com/Ultimaker/Cura/pull/6016

Ghostkeeper, 2019-07-22:
Oops, sorry, no that fix was only for project files, not for imported profiles. Seems we need a different fix for these profiles.

Vandrasc, 2019-07-26:
The fix will be in the 4.2 stable release.

closed 2019-07-26 by Vandrasc

Liger0, 2019-07-26:
It also happens from 4.2 to 4.2. exported profiles are just un-importable no matter where they were generated.

OldGaf, 2019-07-31:
I have 4.2 and have this issue still.

The machine defined in profile E:/Downloads/CR-10-Mech.curaprofile (fdmprinter) doesn't match with your current machine (creality_base), could not import it.

eastabrooka, 2019-08-01:
I have this too, When I try and import https://www.3dprintedtabletop.com/resources/3d-printing-profiles/

OldGaf, 2019-08-01:
This was "fixed" in 4.2.1... but that has introduce more issues with profiles.

e2p2, 2019-08-10:
I just tried upgrading from 4.1.0 to 4.2.1 and none of the Ender 3 profiles migrated over. The profiles exported from 4.1.0 and attempted to import into 4.2.1 resulted in the same issue from the top of the thread.

leoheck, 2019-09-07:
Same here, 

[image]