Ultimaker/Cura #6034: Support angle rotation

tablatronix, 2019-07-20:
I did not see a way to modify the orientation of supports, other than rotating the model.

Would it be possible to rotate the support structure generation so that overhangs bridge better etc.

[image]
[image]

dstulken, 2019-07-21:
This is an aside, but I've always thought a "radial zig-zag" would be a nice support pattern.  Conceptually it would be the opposite of concentric, connected spokes instead of hoops, and would provide the result you (and I) desire for *all* sides of a model.

mbartlett21, 2019-07-22:
@dstulken 

> This is an aside, but I've always thought a "radial zig-zag" would be a nice support pattern. Conceptually it would be the opposite of concentric, connected spokes instead of hoops, and would provide the result you (and I) desire for _all_ sides of a model.

Great idea

fieldOfView, 2019-07-22:
On-topic:

Have you tried the "Infill Line Directions" setting? 

Off-topic:

Wouldn't a radial zigzag as described become sparser and sparser the further away it is from the center?

Ghostkeeper, 2019-07-22:
This is already implemented. The setting is called "Support Infill Line Direction". Here I've set this direction to 30 degrees:
[image]

closed 2019-07-22 by Ghostkeeper

dstulken, 2019-07-23:
> Off-topic: Wouldn't a radial zigzag as described become sparser and sparser the further away it is from the center?

It would.  

As a general solution, I'm picturing rings of triangular (or rather, pie-shaped) zig-zags.  Closest to the model would be a single point, then pie-shaped spokes radial from that.  When the separation became too much, a second (and successive) ring of this shape would be added, with each one's points starting mid-way between the prior ring's zags.    It would not be of uniform density as measured along a ray outward from the model center, but it would still fill in appropriately ensure that no model element had too large of a gap to bridge.

Alternately, you could just accept that as a limitation, and allow the radial pattern to become sparse.  If you need density on the outer edge, you have to deal with a denser center.   For models with comparatively short overhangs on any given side, this would be plenty adequate for a lot less software work, and not a lot of negatives to the user.

Ghostkeeper, 2019-07-23:
I think that the radial pattern could work well with patterns where there is a circle or square that needs overhang. Basically just as nice as regular zig zag then. As soon as you get more complex shapes though, it'll become a terrible pattern. Take this random shape from something that I printed once:

[image]

The centre of the radial pattern will land somewhere between the two brackets. The two brackets themselves and the rings in the back will be supported by a few loose lines that radiate outward from the invisible centre. The bracket in the back will be supported better because it's closer to the centre.

But even for simple round shapes it probably won't be as effective as zigzag, because you can't print it without travel moves and you can't squish the support after printing in order to remove it.

tablatronix, 2019-07-23:
Is this a candidate for tree supports?

Ghostkeeper, 2019-07-24:
Yeah tree support works with the support angle, if that's what you're asking, but it by default has no infill. You can give it infill and then that infill will be influenced by the support angle setting.

I printed that bracket also with tree support, if that was what you were asking.

tablatronix, 2019-07-24:
heh yeah I meant that specific model, I have yet to have to print something that needs elaborate supports to test it, but it does look pretty useful.

Ghostkeeper, 2019-07-25:
We had 3D concentric at some point but removed it (I think in 4.0?) because it wasn't useful at all.

3D concentric had the problem that it could start parts in mid-air. And this happened pretty often, every time the pattern didn't happen to fit well with the diameter of the supported part.