Ultimaker/Cura #6036: Filament Runout Sensor Not working

CGLee65, 2019-07-20:
I am running Cura 4.1.0 and when Printing from the USB the Filament Runout Sensor does not work.  I tried it with printing from the SD Card and it works fine just not when printing from the USB.

Vandrasc, 2019-07-22:
Hi @CGLee65 , could you please use the template so we can gather the necessary information in order to figure this out? 
Thank you!

aquajazz, 2021-02-03:
Have the same issue. The Filament Runout Sensor is working with SD-Card print from Cura, but not with USB-Print via Cura.

Application version
4.8.0
Platform MacOS 11.1 Big Sur
MacBook Pro 2019
2,6 GHz 6-Core Intel i7
16GB of memory

Printer
Creality Ender 5 Plus
Ver. 1.70.1 BL
Silent Board
Creality Metal Extruder
PTFE

Reproduction steps
print stl-file via SD-Card
cut filament just befor sensor
print stl-file via USB
cut filament just before sensor

Actual results
SD-Card: when filament runout sensor light switch off, print-job pauses until I react on the request of filament change on the display. I`m able to resume printing with a new roll of filament.

USB: When the sensor detects, that filment runs out, sensor light switches off. But the print-job do not stop, it is continued without filament.

Expected results for USB print-jobs
If printer runs out of filament:
- message via Cura or printer display
- printer switch to "pause"
- resume print job after the filament change

Project file
Any stl-file sliced by this Cura version.

Thanks,
#Thomas

aquajazz, 2021-02-03:
would be great, if we could reopen the topic

Ghostkeeper, 2021-02-10:
The Cura developers can't implement this though. We don't have your printer. The Ultimaker S5 has a filament flow sensor but doesn't connect via USB (and the pause is properly implemented in the firmware).

Cura will send commands for the printer to execute, filling up the command buffer of the printer (so Cura is usually like 16 commands ahead of the printer). If the printer needs to pause for any reason, it communicates that to Cura by simply not taking any new commands from its buffer, and so the buffer stays full and Cura will simply wait until it gets a response from the printer that it's okay to send the next one over.

It seems that your printer firmware doesn't do that then. Cura will wait for the "ok" message, and apparently your firmware keeps sending "ok" messages and executing the motion commands. There is no way for Cura to know that the filament has ran out then.

That means that there is no way for Cura to know when to pause, so Cura can't pause when the filament ran out. In short, I'm putting the blame on your printer's firmware.
What's more, our developers don't have your printer so we can't test any fixes even if it were possible. I'm afraid we'll have to close this as a won't-fix.

closed 2021-02-10 by Ghostkeeper