Ultimaker/Cura #6037: Cura 4.1 at print time will not use current Custom PLA

Wozillah, 2019-07-21:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

Wozillah, 2019-07-21:
Print job sent to Cura 4.1 is halted and asks for PLA to be changed from the generic PLA in print core 1.
In prepare a custom material has been selected and appears as being used at the top of the screen.
I can find no way to not receive an error that requires me to override the mis-match and print anyway.

Wozillah, 2019-07-21:
Printer is Ultimaker 3

Vandrasc, 2019-07-22:
Hi @Wozillah , could you please share the project file with us (File>Save..) so we can see what goes wrong in your configuration? 

Also, to be sure I understand the problem correctly, the Custom material is selected and not recognized when trying to print with that configuration? 

Thanks!

Wozillah, 2019-07-23:
It is not the project file. I simply selected a custom PLA in preparation and in preferences and those do not show up at print time. When sent to the printer is still using the generic POA and demands that I override that setting I don’t know how to be more clear

Wozillah, 2019-07-23:
I will send a pic

nallath, 2019-07-24:
If you want to embed pictures, you need to post them in the github interface. Sending them as an attachement doesn't work.

Ghostkeeper, 2019-07-24:
I'd like to see your reproduction steps. I guessed these reproduction steps:
1. Create a completely new custom material.
2. Change the material type to Chocolate, the name to Belgian and the colour to White, to be able to recognise it.
3. Add a small cube to the build plate.
4. Send the print to the printer via wifi.

The printer now complains that it doesn't have chocolate loaded in its first extruder, which is expected behaviour.

Wozillah, 2019-07-24:
@Ghostkeeper 
Yes your description of steps matches my experience. I create a custom material in cura and the printer has no idea what that material is. So this is how it is supposed to work or as you say expected. I expected the printer and Cura to be more integrated. Thank you for clearing this up for me.

Ghostkeeper, 2019-07-25:
Well, the printer is complaining that you're trying to print in Chocolate but it still has PLA in its nozzle. So it's asking you to load her up with chocolate, raring to go!

You can ask it to continue printing anyway, of course.

Yes, that is intended behaviour.

closed 2019-07-25 by Ghostkeeper