Ultimaker/Cura #6038: No top layer while reducing section

notmine642, 2019-07-22:
**Cura 4.1.0**

**Win 7 64bit**

**Custom**

**Reproduction steps**
Print a part that has a section that is reduced at mid height

**Actual results**
top layer is not printed and gap appears between inner layer of lower part and outer layer of upper part

**Expected results**
few layers to fill the gap

**Additional information**
[image]
[image]
[image]
[image]

I have 2 wall line count, if I put 3 it is quite ok.
The question is why Cura does not add a full layer to support the upper part ?

Thanks for your help

Ghostkeeper, 2019-07-22:
Reduce the value of the Skin Removal Width if you want it to support the walls on top better. For models with curved lower and upper sides the Skin Removal Width is very useful, but for your case it is detrimental to your part's strength.

closed 2019-07-22 by Ghostkeeper

notmine642, 2019-07-22:
thanks for your help, there is so many parameters in cura, I know the use of one more now !
Best regards